Open-CMSIS-Pack/cpackget #1: rm --purge does not remove all the pack related files

LMESTM, 2021-08-25:
**Describe the bug**
Thanks for this first cpackget version.

My $CMSIS_PACK_ROOT folder contained an old test pack with files like:
 vendor.pack_name.0.0.2.pack
 vendor.pack_name.0.0.2.pdsc
 vendor.pack_name.0.0.2.lic.txt

After command 
./cpackget.exe pack rm --purge vendor.pack_name
the .pack and .pdsc is well removed, but .lic.txt is still there. 

**To Reproduce**
Steps to reproduce the behavior:
1. Install a pack with specific licence file .lic.txt
2. Run the cpackget with the following args...
./cpackget.exe pack rm --purge vendor.pack_name
3. Check if .lic.txt file is well removed or not

**Expected behavior**
I would expect .lic.txt file to be removed 

**Environment (please complete the following information):**
 - OS: [e.g. windows/linux/darwin]
     windows with git bash 
 - Version [e.g. v0.0.3]
    today's main (cpackget -V doesn't return a version as of now)

jkrech, 2021-08-25:
@LMESTM thanks for reporting. This is a product alignment issue. 
I believe you have use MDK pack installer to install the pack. 
The MDK pack installer does keep a copy of the license file in the .Download folder when installing.
cpackget has not implemented the license handling at all yet. I was wondering whether we immediately implement the extended license handling.

chaws, 2022-04-04:
@jkrech Now that cpackget has the extract-license feature. Should `rm --purge` remove license files as well if any?

jkrech, 2022-04-04:
I think the answer is "yes".

chaws, 2022-04-06:
@jkrech Just pushed a PR for fixing this issue. There's one thing to note:

In the cache (.Download) folder, the license file name is appended to the full pack's file name:
* vendor.pack_name.0.0.2.pdsc
* vendor.pack_name.0.0.2.pack
* vendor.pack_name.0.0.2.pack.lic.txt**

So these are going to be purged.

jkrech, 2022-04-06:
Makes sense to me.

closed 2022-04-13 by chaws