Open-CMSIS-Pack/cpackget #19: cpackget shall find download url for public packs

jkrech, 2021-11-05:
Today cpackget requires a download url for downloading and installing a public pack.
Cbuildgen is creating the packlist with download urls for the required packs. 
We should implement the translation from packID to download url in cpackget.

`cpackget pack add ARM.CMSIS.5.7.0`

Note: in case no version is specified the greatest version found is used.

Process:
assumption - file: ARM/CMSIS/5.7.0/ARM.CMSIS.pdsc - does not exist in $CMSIS_PACK_ROOT

```
if exist $CMSIS_PACK_ROOT/.Download/ARM.CMSIS.5.7.0.pack
  cpackget pack add file://..../.Download/ARM.CMSIS.5.7.0.pack
else
  if not exist $CMSIS_PACK_ROOT/.Web/ARM.CMSIS.pdsc
      if not exist $CMSIS_PACK_ROOT/.Web/index.pidx
          cpackget index https://www.keil.com/pack/index.pidx
      read index.pidx find url for ARM.CMSIS.pdsc
      download $url to $CMSIS_PACK_ROOT/.Web/
  read releases section of ARM.CMSIS.pdsc
  find version 5.7.0
  check if  url attribute is set in <release> or if the default url is constructed 
  cpackget pack add $url
```

chaws, 2021-11-05:
Just complementing:

1. cpackget should always construct the final pack file name base on packID attributes, this is necessary when packs from from github artifacts links, e.g. v0.0.1.zip 
2. cpackget will look into a pack's decompressed folder searching for the pdsc file. If the file isn't there then
2.1. If there's just a single folder in the pack's content AND if the pdsc is contained there, cpackget will use the pdsc's dirname as the pack root folder, decompressing it into the CMSIS_PACK_ROOT/Vendor/PackName/x.y.z/
2.2. If no pdsc is present or if there are more than just one folder in the pack file, abort and raise an error

chaws, 2021-11-08:
Hi @jkrech 

I think we should extend the scenarios of "pack add" to the following (I copied most of your code snippet)
```
 1. snippet for "cpackget pack add ARM.CMSIS[.5.7.0]"
 2. packId = ARM.CMSIS[.5.7.0]
 3. if packId.version is available && exist $CMSIS_PACK_ROOT/.Download/ARM.CMSIS.5.7.0.pack
 4.     cpackget pack add file://..../.Download/ARM.CMSIS.5.7.0.pack
 5.     exit 0
 6. 
 7. if NOT exist $CMSIS_PACK_ROOT/.Web/index.pidx:
 8.     cpackget index https://www.keil.com/pack/index.pidx
 9.
10. # Make up pack url from pdsc tag's URL attribute + packId
11. find pdscTag matching ARM.CMSIS[.5.7.0] # if packId.version is not present, jump to line 14
12. pack_url = pdscTag.URL + packId
13. if pack_url returns HTTP 404:
14.     if NOT exist $CMSIS_PACK_ROOT/.Web/ARM.CMSIS.pdsc:
15.         read index.pidx find url for ARM.CMSIS.pdsc
16.         download ARM.CMSIS.pdsc to $CMSIS_PACK_ROOT/.Web/
17.     read releases section of ARM.CMSIS.pdsc # latest or packId.version
18.     if releaseTag does NOT contain url attribute:
19.         raise error "cannot find pack URL"
20.         exit -1
21.     pack_url = releaseTag.url
22. cpackget pack add $url
```

I think it's a good idea to try to make the pack's url using the URL attribute already present in $CMSIS_PACK_ROOT/.Web/index.pidx as it is the most common scenario. Only if the pack's url returns 404 we should look for the url inside releases tag in the pack's pdsc file. What do you think?

Oh, and I have another question:
While looking at [index.pidx](https://sadevicepacksprodus.blob.core.windows.net/idxfile/index.pidx) I noticed "deprecated" in some pdsc tags. What should we do if the pdsc tag used to find the pack's pdsc in index.pidx has `deprecated="..."` in it? Should cpackget raise a warning about that? What happens when both "deprecated" and "replacement" are present in the pdsc tag we're downloading? I know this is a corner case, but it would be nice to predict the behavior for these scenarios as well.

jkrech, 2021-11-09:
Good points.
I like the idea of trying to download a pack version based on the url provided by the index.pidx (assuming that pdsc and pack version files are located in the same web folder). Please keep in mind that in the past we had issues to clearly identify that the download failed as websites not always respond with 404. This means you will need to check that a valid pack is downloaded, but that's what you do anyhow, right?

If you see the "deprecated" and/or "replacement" attributes it makes sense to issue a warning message. 
In my opinion we must not try to be "too smart" and install the replacement pack instead or in addition.

chaws, 2021-11-09:
Yes, I haven't added all possible checks in the snippet as it would cover a lot of basic error checks, like 404s or malformed PDSC files.

Ack regarding "deprecated"/"replacement".

I'll start working on this asap, as the [integration PR](https://github.com/Open-CMSIS-Pack/devtools/pull/50) is almost ready.

closed 2021-11-12 by jkrech