Open-CMSIS-Pack/cpackget #33: cpackget pack rm not working with specific versioning

fred-r, 2022-03-10:
cmsis-toolbox 0.9.1 

Issue with pack versioning ?

```
$ cpackget pack rm STMicroelectronics::Example_HAL_GPIO_Toggle@0.0.1-alpha1
E: bad pack name: pack version should be versioned like 0.0.0, and optionally have a suffix containing letters
```

fred-r, 2022-03-10:
@jkrech @brondani : I do not know if this is ok to raise tickets here for cmsis toolbox when cpackget is involved ?

jkrech, 2022-03-10:
@fred-r , yes it is good to raise issues against cpackget in this dedicated repository.
Please note that at the point cpackget command line was developed, we had not agreed on this new syntax for the packID.
Therefore your command should read:
```
cpackget pack rm STMicroelectronics.Example_HAL_GPIO_Toggle.0.0.1-alpha1
```
Obviously we need to get this aligned at some point.

chaws, 2022-03-11:
@jkrech is this something to be added to cpackget pack version parser?

jkrech, 2022-03-11:
Any "version" of a pack that can be installed also needs to be removed by cpackget. In order to do so any legal version (semver.org) can be specified here.

So any packID listed with this command:
```
$cpackget pack list
```
Should also work for removing a pack
```
$cpackget pack rm <packID>
```

chaws, 2022-03-11:
@jkrech That is already possible:
```bash
$ wget -q https://github.com/Open-CMSIS-Pack/cpackget/releases/download/v0.3.0/cpackget_0.3.0_linux_amd64.tar.gz

$ tar -xf cpackget_0.3.0_linux_amd64.tar.gz

$ cd cpackget_0.3.0_linux_amd64/

$ export CMSIS_PACK_ROOT=my-pack-root

$ ./cpackget init https://keil.com/pack/index.pidx
I: Using pack root: "my-pack-root"
W: "my-pack-root/.Web/index.pidx" not found. Creating a new one.
I: Overwriting public index file my-pack-root/.Web/index.pidx
I: Downloading index.pidx 100% |██████████████████████████████████████████████████████████████████████████████████████████████████████████| (98/98 kB, 137.732 kB/s)

$ ./cpackget pack add NordicSemiconductor.nRF_Libraries.4.0.0-2.alpha
I: Using pack root: "my-pack-root"
I: Downloading NordicSemiconductor.nRF_Libraries.pdsc 100% |████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (84/84 kB, 135.985 kB/s)        
I: Adding pack "NordicSemiconductor.nRF_Libraries.4.0.0-2.alpha"
I: Downloading NordicSemiconductor.nRF_Libraries.4.0.0-2.alpha.pack 100% |████████████████████████████████████████████████████████████████████████████████████████████████████| (293/293 kB, 178.421 kB/s)        
I: Extracting files to my-pack-root/NordicSemiconductor/nRF_Libraries/4.0.0-2.alpha 100% |█████████████████████████████████████████████████████████████████████████████████████████| (167/167, 13835 it/s)

$ ./cpackget pack list
I: Using pack root: "my-pack-root"
I: Listing installed packs
I: NordicSemiconductor.nRF_Libraries.4.0.0-2.alpha

$ ./cpackget pack rm NordicSemiconductor.nRF_Libraries.4.0.0-2.alpha
I: Using pack root: "my-pack-root"
I: Removing [NordicSemiconductor.nRF_Libraries.4.0.0-2.alpha]

$ ./cpackget pack list
I: Using pack root: "my-pack-root"
I: Listing installed packs
I: (no packs installed)
```

My question was about adopting the syntax @fred-r attempted with "::" instead of ".", or should cpackget accept both?

jkrech, 2022-03-11:
I think we should discuss this first.

chaws, 2022-03-14:
Yeah, I think supporting both formats would be great for backwards compatibility in cpackget. 

Additionally, adopting the ">=" syntax might be a good thing adding packs, e.g. `cpackget pack add ARM::CMSIS>=5.5.0`, what do you think?

chaws, 2022-03-25:
Hi @jkrech, I have added support for [parsing pack names using `::@`](https://github.com/Open-CMSIS-Pack/cpackget/pull/41) as well as the current syntax. Both formats will be supported. But the new syntax brings some extra complexity that I believe needs extra discussion.

I'll start off this discussion with the following use case:

For started, let's assume `ARM.CMSIS` has available versions `5.8.0`, `5.7.0`, and `5.6.0`, then I'm assuming the following:

* `cpackget pack add ARM::CMSIS@5.7.0`: installs `5.7.0`
* `cpackget pack add ARM::CMSIS>=5.6.0`: noop because `5.7.0` is already installed satisfies the condition
* `cpackget pack add ARM::CMSIS`: should it look for whatever version installed, e.g. `5.7.0` or should it ALWAYS look for latest one, e.g. `5.8.0`?

Also, should `>=` syntax be allowed when removing packs? If so, what would be the outcome of that?

jkrech, 2022-03-25:
@chaws, these are very good questions and from my point of view you went a step further than I had initially thought, which is clearly my mistake.
In a first step I was just thinking of replicating what we already had using the 'dot' separated notation of a packID:
```
<vendor>.<packname>   e.g. ARM.CMSIS -> this is always implying the latest version you can find in the downloaded pdsc
<vendor>.<packname>.<version> e.g. ARM.CMSIS.5.7.0 -> this requires an exact match otherwise report error
```
This would then map to the YML syntax
```
<vendor>::<packname>
<vendor>::<packname>@<version>
```
The above should be the straight forward part.
Now to your scenario above
a) Agreed as it falls into the straight forward mapping
b) Agreed as well
c) In this case no version is specified, hence any version installed satisfies this request. It does not require the latest version
However wanting to update to the latest version is an extremely common use case which we have not covered in csolution today, basically a project requiring the system to install the latest version of a pack whenever it is build. Not sure this is a useful feature.
For a package installer this is different. So I would propose something like:
```
$cpackget pack add ARM::CMSIS@latest
```
Another idea we could steal from python's pip install is the '~=' requesting the latest compatible version.
Assume that there is the following pack versions of ARM.CMSIS available: 5.6.0, 5.7.0, 5.8.0 and 6.0.0
```
$cpackget add ARM::CMSIS@~5.7.0
```
The above would imply that 6.0.0 is not compatible (Semantic Versioning) therefore the latest compatible version gets installed, regardless which versions are already installed.

I think we agree that for  "add" we will only ever install a maximum of one pack version.
I am unsure whether we want to allow removing multiple pack versions?
But the outcome of 
```
cpackget rm ARM::CMSIS>5.6.0
```
would result in both5.7.0 and 5.8.0 being removed.

chaws, 2022-04-01:
Hi @jkrech I pushed new changes to the PR. Here are some attempts I've tried:

```bash
# Setup
[0|0]name-syntax $ export CMSIS_PACK_ROOT=packroot
[0|0]name-syntax $ cpackget init https://keil.com/pack/index.pidx
I: Using pack root: "packroot"
W: "packroot/.Web/index.pidx" not found. Creating a new one.
I: Overwriting public index file packroot/.Web/index.pidx
I: Downloading index.pidx 100% |███████████████████████████████████████████| (99/99 kB, 172.992 kB/s)        
[0|0]name-syntax $ cpackget pack list
I: Using pack root: "packroot"
I: Listing installed packs
I: (no packs installed)

# Adding any version without any previously installed targets latest available
[0|0]name-syntax $ cpackget pack add "ARM::CMSIS"
I: Using pack root: "packroot"
I: Downloading ARM.CMSIS.pdsc 100% |█████████████████████████████████████| (238/238 kB, 165.929 kB/s)        
I: Adding pack "ARM::CMSIS"
I: Downloading ARM.CMSIS.5.8.0.pack 100% |██████████████████████████████████| (34/34 MB, 46.209 MB/s)        
I: Extracting files to packroot/ARM/CMSIS/5.8.0 100% |███████████████████████| (8133/8133, 9921 it/s)        

# Adding a specific version with "@"
[0|0]name-syntax $ cpackget pack add "ARM::CMSIS@5.0.0"
I: Using pack root: "packroot"
I: Adding pack "ARM::CMSIS@5.0.0"
I: Downloading ARM.CMSIS.5.0.0.pack 100% |██████████████████████████████████| (93/93 MB, 8.997 MB/s)         
I: Extracting files to packroot/ARM/CMSIS/5.0.0 100% |███████████████████████| (4439/4439, 2265 it/s)        

# Adding a version with minimum compatible version available
[0|0]name-syntax $ cpackget pack add "ARM::CMSIS@~4.4.0"
I: Using pack root: "packroot"
I: Adding pack "ARM::CMSIS@~4.4.0"
I: Downloading ARM.CMSIS.4.5.0.pack 100% |██████████████████████████████████| (97/97 MB, 46.836 MB/s)        
I: Extracting files to packroot/ARM/CMSIS/4.5.0 100% |███████████████████████| (4029/4029, 1996 it/s)        

# Listing all installed packs
[0|0]name-syntax $ cpackget pack list
I: Using pack root: "packroot"
I: Listing installed packs
I: ARM.CMSIS.4.5.0
I: ARM.CMSIS.5.0.0
I: ARM.CMSIS.5.8.0

# Attempt to install ">=5.5.0" is noop because 5.8.0 is present
[0|0]name-syntax $ cpackget pack add "ARM::CMSIS>=5.5.0"
I: Using pack root: "packroot"
I: Adding pack "ARM::CMSIS>=5.5.0"
E: pack ARM::CMSIS>=5.5.0 is already installed here: packroot/ARM/CMSIS/5.8.0

# Removing specific version
[255|0]name-syntax $ cpackget pack rm "ARM::CMSIS@5.8.0"
I: Using pack root: "packroot"
I: Removing [ARM::CMSIS@5.8.0]

[0|0]name-syntax $ cpackget pack list
I: Using pack root: "packroot"
I: Listing installed packs
I: ARM.CMSIS.4.5.0
I: ARM.CMSIS.5.0.0

[0|0]name-syntax $ cpackget pack add "ARM::CMSIS@latest"
I: Using pack root: "packroot"
I: Adding pack "ARM::CMSIS@latest"
I: Extracting files to packroot/ARM/CMSIS/5.8.0 100% |██████████████████████| (8133/8133, 10029 it/s) 
```

jkrech, 2022-04-04:
@fred-r @brondani  and others, this is looking good to me but maybe you disagree or think we are missing out on scenarios.

closed 2022-04-04 by chaws

chaws, 2022-04-04:
Thanks @jkrech ! I have merged the patch because it was a big one. I'll tackle other open issues now, but if there's need for adding more changes to this one, please let me know

brondani, 2022-04-04:
It looks good to me. Removing "pack ranges" (@~ or >=) seems to be less relevant, but removing all installed versions of a given pack sounds useful, for example `cpackget pack rm "ARM::CMSIS"` without specifying the version. I believe this use case is already supported, is it right @chaws?

chaws, 2022-04-04:
@brondani yes!

```bash
$ export CMSIS_PACK_ROOT=../packroot/
$ cpackget pack list
I: Using pack root: "../packroot"
I: Listing installed packs
I: ARM.CMSIS.5.8.0

$ cpackget pack add ARM::CMSIS@4.5.0
I: Using pack root: "../packroot"
I: Adding pack "ARM::CMSIS@4.5.0"
I: Downloading ARM.CMSIS.4.5.0.pack 100% |███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (97/97 MB, 44.106 MB/s)        
I: Extracting files to ../packroot/ARM/CMSIS/4.5.0 100% |█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (4029/4029, 2030 it/s)        

$ cpackget pack add ARM::CMSIS@4.4.0
I: Using pack root: "../packroot"
I: Adding pack "ARM::CMSIS@4.4.0"
I: Downloading ARM.CMSIS.4.4.0.pack 100% |███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (97/97 MB, 3.571 MB/s)         
I: Extracting files to ../packroot/ARM/CMSIS/4.4.0 100% |█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (4027/4027, 2021 it/s)        

$ cpackget pack list
I: Using pack root: "../packroot"
I: Listing installed packs
I: ARM.CMSIS.4.4.0
I: ARM.CMSIS.4.5.0
I: ARM.CMSIS.5.8.0

$ cpackget pack rm ARM::CMSIS@4.5.0
I: Using pack root: "../packroot"
I: Removing [ARM::CMSIS@4.5.0]

$ cpackget pack list
I: Using pack root: "../packroot"
I: Listing installed packs
I: ARM.CMSIS.4.4.0
I: ARM.CMSIS.5.8.0
I: AWS.coreHTTP.pack

$ cpackget pack rm ARM::CMSIS
I: Using pack root: "../packroot"
I: Removing [ARM::CMSIS]

$ cpackget pack list
I: Using pack root: "../packroot"
I: Listing installed packs
```