Open-CMSIS-Pack/cpackget #34: cpackget pack list command stops when an incorrect pack is detected

ReinhardKeil, 2022-03-15:
When using the `cpackget pack list` command it appears to stop on the first pack that uses inconsistent naming conventions.
```txt
cpackget pack list
I: Using pack root: "C:\Keil_v5\ARM\PACK"
I: Listing installed packs
I: AmbiqMicro.Apollo_DFP.1.2.0
I: Arm-Packs.PKCS11.1.0.0
I: ARM.AMP.1.1.0
I: ARM.CMSIS-Driver.2.6.0
I: ARM.CMSIS-Driver.2.6.1
I: ARM.CMSIS-Driver_Validation.2.0.0
I: ARM.CMSIS-FreeRTOS.10.0.1
I: ARM.CMSIS-FreeRTOS.10.3.0
I: ARM.CMSIS-FreeRTOS.10.3.1
I: ARM.CMSIS-FreeRTOS.10.4.3
I: ARM.CMSIS-FreeRTOS.10.4.6
I: ARM.CMSIS.5.7.0
I: ARM.CMSIS.5.8.0
I: ARM.CMSIS_ARMv81M.1.0.0
E: Pack version "0.0.1.20210416" does not match ^(0|[1-9][0-9]*)\.(0|[1-9][0-9]*)\.(0|[1-9][0-9]*)(-(0|[1-9][0-9]*|[0-9]*[a-zA-Z-][0-9a-zA-Z-]*)(\.(0|[1-9][0-9]*|[0-9]*[a-zA-Z-][0-9a-zA-Z-]*))*)?(\+[0-9a-zA-Z-]+(\.[0-9a-zA-Z-]+)*)?$
E: A pack in the cache folder has malformed pack name: ARM.ethos-u-core-driver.0.0.1.20210416.pack
E: unknown behavior
```

I would also suggest to use a unified error messaging format as proposed here: https://github.com/Open-CMSIS-Pack/Open-CMSIS-Pack-Spec/issues/59

The output could read:
```
CMSIS_PACK_ROOT="C:\Keil_v5\ARM\PACK"
AmbiqMicro.Apollo_DFP.1.2.0
Arm-Packs.PKCS11.1.0.0
ARM.AMP.1.1.0
ARM.CMSIS-Driver.2.6.0
ARM.CMSIS-Driver.2.6.1
ARM.CMSIS-Driver_Validation.2.0.0
ARM.CMSIS-FreeRTOS.10.0.1
ARM.CMSIS-FreeRTOS.10.3.0
ARM.CMSIS-FreeRTOS.10.3.1
ARM.CMSIS-FreeRTOS.10.4.3
ARM.CMSIS-FreeRTOS.10.4.6
ARM.CMSIS.5.7.0
ARM.CMSIS.5.8.0
ARM.CMSIS_ARMv81M.1.0.0
ARM.ethos-u-core-driver.0.0.1.20210416.pack - error: pack version incorrect format
ARM.nextpack.1.0.0
   :
info: 1 error(s) detected
```

chaws, 2022-03-22:
Hi @ReinhardKeil Thanks for raising the ticket.
I didn't expect packs to not follow the naming pattern, hence it breaking on the first error there. I'll get a fix to this.

jeromecoutant, 2022-03-25:
Hi

It seems that `cpackget pack list` doesn't work?

I consider my pack root clean...
And I got the same error message

```
D: Extracting pack info from "C:.pack_root.STMicroelectronics.board_resources_B-U585I-IOT02A.0.0.2.pack"
E: Pack vendor "" does not match ^[0-9a-zA-Z_\-]+$
E: A pack in the cache folder has malformed pack name: C:.pack_root.STMicroelectronics.board_resources_B-U585I-IOT02A.0.0.2.pack
```

jkrech, 2022-03-25:
Your problem sounds like issue #36 reported earlier.
This issue got fixed in https://github.com/Open-CMSIS-Pack/cpackget/releases/tag/v0.3.2
This version/fix will be included in the next cmsis-toolbox release next week.

jeromecoutant, 2022-03-25:
> This version/fix will be included in the next cmsis-toolbox release next week.

I confirm
Thx

closed 2022-04-14 by jkrech