Open-CMSIS-Pack/cpackget #35: Suggestion: Improve usiblity

ReinhardKeil, 2022-03-15:
The `cpackget` tool is already quite good and **this is a proposal** to improve the usability.  

I have not fully understood the `PDSC` command.  How does this differ from the `PACK` command?

I suggest to simplify the command entry by:
  - accept a single character as alternative
  - use separate commands for `pack`

Usage could look like:

```
Usage:
  cpackget [command] [flags]

command:
  add         Download or install a pack to local pack root
  init          Initialize or update the local pack root
  help        Help about any command
  list           List all packs
  remove   Delete a pack from local pack root
  purge      Remove a pack from local pack root and download cache

flags:
  -h, --help               help for cpackget
  -R, --pack-root string   Specifies pack root folder. Defaults to CMSIS_PACK_ROOT environment variable (default "C:\\Keil_v5\\ARM\\PACK")
  -q, --quiet              Run cpackget silently, printing only error messages
  -v, --verbose            Sets verboseness level: None (Errors + Info + Warnings), -v (all + Debugging). Specify "-q" for no messages
  -V, --version            Prints the version number of cpackget and exit

Use "cpackget [command] --help" for more information about a command.
```

For example the `cpackget list` command could be written as `cpackget l`.  It could show all available packs from the `index.pidx` file and indicate that packs are installed with ` - installed`.  It could also have the option to filter, for example with:

 - `cpackget list arm`  show all packs from Arm
 - `cpackget list *st*` show all packs that include the letters `st` in the file name

The `cpackget add` command could automatically download the pack, if it is available in the `index.pidx` file from the host service, this would reduce the command to:
 - `cpackget add ABOV.A31G21x_Series.1.5.1.pack`

jkrech, 2022-03-15:
The pdsc command allows to add and remove "local repositories", which are stored in $CMSIS_PACK_ROOT/.Local/local_repository.pidx
Therefore it does neither download nor extract any pack content.
If we remove the pack vs. pdsc command the tool can simply look at the specified file extension. If the user specifies a pdsc file it is about adding or removing as pdsc from the local_repository.pidx otherwise it is a packfile (*.pack) or a packID (no extension)

```cpackget add ABOV.A31G21x_Series.1.5.1.pack```
would actually assume that the actual pack file is present in the current directory and try to install it. 
Similar to this use case:
```cpackget add ./myPackScratch/ABOV.A31G21x_Series.1.5.1.pack```

Instead you need to specify the pack with the 'packID' (without file extension)
```cpackget add ABOV.A31G21x_Series.1.5.1```

In order to install the latest available version you can skip the version in the  ID
```cpackget add ABOV.A31G21x_Series```

ReinhardKeil, 2022-03-15:
Thanks for explaining. I would suggest to improve the command specific help pages as this type of information should be displayed when you enter `cpackget help add` for example.

ReinhardKeil, 2022-03-15:
I noted also that the current help output of the tool is somewhat incomplete and does not reflect the current implementation.

chaws, 2022-03-15:
Hi @ReinhardKeil , Thanks for your suggestion!

Which `help` command have you attempted? Could you please try running the following?

1. Help for `pack add`:
```bash
$ cpackget pack add --help
Adds a pack using the file specified in "<pack path>" or using packs URLs provided by "-f <packs list>".
The file can be a local file or a file hosted somewhere else on the Internet.
If it's hosted somewhere, cpackget will first download it then extract all pack files into "CMSIS_PACK_ROOT/<vendor>/<packName>/<x.y.z>/"
If "-f" is used, cpackget will call "cpackget pack add" on each URL specified in the <packs list> file.

Usage:
  cpackget pack add [<pack path> | -f <packs list>] [flags]

Flags:
  -a, --agree-embedded-license       agrees with the embedded license of the pack
  -x, --extract-embedded-license     extracts the embedded license of the pack and aborts the installation
  -h, --help                         help for add
  -f, --packs-list-filename string   specifies a file listing packs urls, one per line
...
```
2. Help for `pdsc add`:
```bash 
$ cpackget pdsc add --help
Adds a pack via pdsc file specified in <path/to/TheVendor.ThePack.x.y.z.pdsc>.
cpackget will add the pdsc entry to CMSIS_PACK_ROOT/.Local/local_repository.pidx.

Usage:
  cpackget pdsc add <path/to/TheVendor.ThePack.x.y.z.pdsc> [flags]
...
```

From your suggestion above, I see that you unified `pdsc` and `pack`, thus having `add/rm/list/purge` as a direct subcommand of `cpackget`, correct? It would definitely simplify it. @jkrech what are your thoughts?

If we opt for unifying sub-command like that, perhaps we could have several ways of adding packs, including local ones:
1. `cpackget add ARM.CMSIS`: search index.pidx, download and install
2. `cpackget add ARM.CMSIS.5.5.0`: search index.pidx, download and install
3. `cpackget add ARM.CMSIS.5.5.0.pack`: install only
4. `cpackget add ARM.CMSIS.5.5.0-alpha.pdsc`: do not install, add it to `local_repository.pdsc` only

This would overload the `add` command a little, but apparently 1, 2 and 3 are the most used ways anyway?

ReinhardKeil, 2022-03-15:
@chaws I was perhaps using an older version.  But you are right, I suggest to combine the PDSC and PACK command.

closed 2022-05-04 by jkrech