Open-CMSIS-Pack/cpackget #39: [questions] cpackget init

fred-r, 2022-03-16:
According to the help:
```
$ cpackget init --help
Initializes a pack root folder specified by -R/--pack-root command line
or via the CMSIS_PACK_ROOT environment variable with the following contents:
  - .Download/
  - .Local/
  - .Web/
  - .Web/index.pidx (downloaded from <index-url>)
```

But:
```
$ cpackget init -R ./packroot/
I: Using pack root: "./packroot/"
W: "packroot\.Web\index.pidx" not found. Creating a new one.

$ tree -a
.
|-- .Download
|-- .Local
`-- .Web
    `-- index.pidx

3 directories, 1 file

$ cat .Web/index.pidx
<?xml version="1.0" encoding="UTF-8"?>
<index schemaVersion="1.1.0">
 <vendor>packroot\.Web\index</vendor>
 <url></url>
 <pindex></pindex>
</index>

```

[Q1] .Web/index.pidx ==> where is it downloaded from ? What is <index-url> ?

Then we can install a pack:
```
$ cpackget pack add https://xxxxx/STMicroelectronics.STM32U5xx_DFP.0.1.0-alpha1.pack -R .
I: Using pack root: "."
I: Adding pack "STMicroelectronics.STM32U5xx_DFP.0.1.0-alpha1"
I: Downloading STMicroelectronics.STM32U5xx_DFP.0.1.0-alpha1.pack 100% || (2.2/2.2 MB, 1.414 MB/s)
I: Extracting files to STMicroelectronics\STM32U5xx_DFP\0.1.0-alpha1 100% || (49/49, 185 it/s)
```

And we end up with : 
```
$ tree -a
.
|-- .Download
|   |-- STMicroelectronics.STM32U5xx_DFP.0.1.0-alpha1.pack
|   `-- STMicroelectronics.STM32U5xx_DFP.0.1.0-alpha1.pdsc
|-- .Local
|   `-- STMicroelectronics.STM32U5xx_DFP.pdsc
|-- .Web
|   `-- index.pidx
|-- STMicroelectronics
|   `-- STM32U5xx_DFP
|       `-- 0.1.0-alpha1
|           |-- Flash
|           |-- Include
|           |-- STMicroelectronics.STM32U5xx_DFP.pdsc
|           `-- Source
`-- pack.idx

```
[Q2] Why this architecture with files duplication ?
The .pack is kept in .Download, why ?
The .pdsc is present in multiple places, why ? Why in .local especially ?

Also:
```
$ cat pack.idx

```

[Q3] What is pack.idx ? Why is it empty ?

jkrech, 2022-03-16:
[Q1] I am not sure whether it is intentional that the 'init' command is not enforcing to specify the download location for the public pack index file. According to the usage: the index-url is optional
```cpackget init [--pack-root <pack root>] [index-url] [flags]```
The public pack index maintained by Arm is located here:
```https://www.keil.com/pack/index.pidx```

```cpackget init --pack-root ./mypackroot https://www.keil.com/pack/index.pidx```

[Q2] Duplication in .download is just a download cache. This allows you to install/uninstall a pack without the need to download it again.
Duplication of the pdsc file into .local is done for all packs that get installed that are not listed in the .Web/index.pidx (public packs). When checking for updates the <url> tag in the pdsc file stored in the .local folder are used to copy a new version of the pdsc to see whether there is new pack versions available. This allows to have PLM of "internal" / "local" packs. You need to install a pack once and can then check for updates of it.

[Q3] The file pack.idx is used by the IDE to find out whether packs have been added/removed. In this case the IDE will reread the  list of installed packs.

[Q4] In order for different tools sharing a single CMSIS-PACK repository we should all agree on the same folder structure.
I believe that standalone/self-contained projects do not require any of these additional folders (.Web/, .Local/, .Download, pack.idx) as these are considered local pack version folders. This means you simply reference a project relative pdsc file. If you were to update this pack version to a newer version, you would need to delete the containing folder and replace it with a copy of the extracted new pack version. This functionality is not supported by cpackget.

jkrech, 2022-03-16:
The file format is clearly something we can look at and the current implementation was driven by CMSIS-Build.
$cbuildgen packlist
command was developed write a file containing download url that would be read by a shell script (cp_install.sh) to do the actual download and unzip (<project>.cp_install)
We replaced the shell script more recently with cpackget while maintaining the filename and format (1 url per line)

Did you know that you can simply pass a file with a list of packIDs (<packvendor>.<packname>)  and the above command would install the latest version of these.
So rather than supporting an xml format as input file to cpackget you should write a script to extract just vendor + name from an pidx formatted file and pass that to the cpackget.

BTW: I cannot think of many circumstances I would want to download all available packs. We are doing it for the online tools and it is huge.

jeromecoutant, 2022-03-16:
> BTW: I cannot think of many circumstances I would want to download all available packs. We are doing it for the online tools and it is huge.

I was thinking about ACI tests

jkrech, 2022-03-22:
@jeromecoutant do you see a strong requirement for cpackget to support pidx as input format?
Or do you think that cpackget should even have a command that creates and maintains a CMSIS-PACK folder with all published packs installed (either latest versions only or even all available versions?).

I personally think this is not the most typical use case and could be achieved by some scripting relatively easily using the existing features.

jeromecoutant, 2022-03-22:
> and could be achieved by some scripting relatively easily using the existing features.

I agree, and already partially done on my side :-)

jkrech, 2022-03-22:
@fred-r are you happy with the answers provided and can we close the issue?

fred-r, 2022-03-22:
Hi @jeromecoutant  and @jkrech ,

regarding the point raised by Jérôme : to me OpenCMSIS aims at providing the vanilla tooling for using the Open-CMSIS-pack-spec concepts.

So, I am more in favor of "atomic services", then the IDE or ACI using it can implement the exact required "advanced behavior".
If we implement to high level services the chances are high that it does not match the exact expectations of the users and that the feature becomes obsolete.

For the other questions: ok, I understand the structure now.
I think it makes sense for an IDE (what do you think @slhultgren @VGRSTM ?).

But, for a "standalone project" with local copies of packs, it might be overkill.
I do not know how you plan to address this point ?
Will it be in the scope of cpackget?

jkrech, 2022-03-22:
@fred-r the cpackget currently only deals with the CMSIS_PACK_ROOT folder and this is how I currently intend to keep it and have a clear separation of concern.
Depending on the use cases we should think about a utility that allows to "privatize" specific or all packs used by a solution / project / layer etc.  From my perspective this should be a copy operation from CMSIS_PACK_ROOT/<packid> -> project destination -> adding PackID and path to the csolution.
Could you raise a new issue for this topic in Open-CMSIS-Pack-Spec and how you anticipate processing this?
We should close the issue here. Would you agree?

jkrech, 2022-03-22:
[Q3] pack.idx (sorry, bad name for historic reasons) is 'touched' by cpackget whenever a pack was added or removed. A running IDE session may decide based on this "activity" to reread the installed packs. E.g. the user may have installed a more recent version of a pack and the effective component selection may change to a new version of the component from this new pack.
The more critical scenario is, in case a pack being used by a project currently loaded in the IDE gets removed and as a result running e.g. a build of the project is going to fail.

fred-r, 2022-03-22:
Hi Joachim,

I worked with cpackget and uvision but this is not what I observe:
```
$ tree -L 2
.
|-- ARM
|   |               `-- CMSIS
|   |               `-- CMSIS-Driver
|   |               `-- PSA
|                   `-- mbedTLS
|-- Keil
|   |               `-- ARM_Compiler
|   |               `-- MDK-Middleware
|   |               `-- NUCLEO-L552ZE-Q_BSP
|   |               `-- STM32F4xx_DFP
|   |               `-- STM32H7xx_DFP
|   |               `-- STM32L4xx_DFP
|   |               `-- STM32L552E-EV_BSP
|   |               `-- STM32L562E-DK_BSP
|   |               `-- STM32L5xx_DFP
|   |               `-- STM32L5xx_TFM-PF
|   |               `-- STM32NUCLEO_BSP
|                   `-- STM32U5xx_DFP
|-- STMicroelectronics
|   |               `-- Example_HAL_GPIO_Toggle
|   |               `-- STM32U5xx_DFP
|   |               `-- STM32U5xx_Drivers
|                   `-- board_resources_B-U585I-IOT02A
`-- pack.idx

23 directories, 1 file

frq09468@LMECWL0888 MINGW64 /c/CMSIS_PACKS
$ cat pack.idx

frq09468@LMECWL0888 MINGW64 /c/CMSIS_PACKS
```

On my side, pack.idx is empty.

jkrech, 2022-03-28:
Any further questions, before we can close this issue?
@fred-r @VGRSTM

VGRSTM, 2022-03-28:
I'm fine to close. Thanks.

closed 2022-03-28 by jkrech