Open-CMSIS-Pack/cpackget #42: cpackget does not accept CRLF in .pdsc ?

fred-r, 2022-04-01:
> cpackget.exe pdsc add C:/Packs/STMicroelectronics.STM32U5xx_Drivers.0.1.1.pack
I: Using pack root: "C:\CMSIS_PACKS"
I: Adding pdsc
E: XML syntax error on line 1: illegal character code U+0003

pdsc which is not accepted:
```
<?xml version="1.0" encoding="UTF-8"?> CR LF
```

pdsc which is accepted:
```
<?xml version="1.0" encoding="UTF-8"?> LF
```

chaws, 2022-04-01:
Hi @fred-r

That's an interesting one. I'd think that the golang stdlib for XML unmarshaling would've handle that, but apparently not.
I'll investigate it.

chaws, 2022-04-04:
Hi @fred-r could you please point me the snippet of this pdsc file you hit the bug?

I have tried using `\r\n` and it worked just fine, so this is not related to CR/CF.

It seems that the error your hit is due to Unicode 0003 character[1], which seems it's character that was added by accident maybe when hitting ctrl+c?

[1] https://www.compart.com/en/unicode/U+0003
[2] https://en.wikipedia.org/wiki/End-of-Text_character

fred-r, 2022-04-06:
Anyway, I think the problem might be that I used pdsc instead of pack command ?

fred-r, 2022-04-06:
Indeed : my bad, it works with the pack command, sorry :-(

closed 2022-04-06 by fred-r

chaws, 2022-04-06:
Oh, thanks for checking, @fred-r !
I'm working on a patch to combine remove `pack` and `pdsc` commands, so you'll only be handling packs with `cpackget add [pack]`, where `pack` can be either a pack file, a pdsc file or simply the pack id