Open-CMSIS-Pack/cpackget #43: Issue to update a local pack ?

fred-r, 2022-04-04:
trying this:
```
> cpackget.exe pack add C:/Packs/STMicroelectronics.STM32U5xx_DFP.0.1.1.pack
I: Using pack root: "C:\CMSIS_PACKS"
E: The pack's pdsc (C:\CMSIS_PACKS\.Local\STMicroelectronics.STM32U5xx_DFP.pdsc) has no release tag matching version "0.1.1"
E: pack version not found in the pdsc file
```

I have this in my local pack's pdsc:

	<releases>
		<release version="0.1.0-alpha1" date="2022-03-09">Add U5_GFX and U5_4M devices</release>
		<release version="0.0.1" date="2021-03-01">Initial release of STM32U5 Device Family Pack.</release>
	</releases>

chaws, 2022-04-04:
Hi @fred-r , shouldn't there be a `0.1.1` version on your pdsc file? I only see `0.1.0-alpha1` and `0.0.1`.

fred-r, 2022-04-04:
Yes, that's the content of the local pack's pdsc, but the content of the .pack's pdsc is like this:

```
  <releases>
    <release version="0.1.1" date="2022-03-17">
      Add component for TIMER code generation
    </release>
    <release version="0.1.0-alpha1" date="2022-03-08">
      Add condition for devices supporting GFX
    </release>
    <release version="0.0.1-alpha1" date="2022-03-01">
      Initial release with HAL/LL drivers
    </release>
  </releases>
```

Looks like the local one takes precedence over the one I try to install.

jkrech, 2022-04-05:
```
cpackget.exe pack add C:/Packs/STMicroelectronics.STM32U5xx_DFP.0.1.1.pack
```
I am struggling to understand the discussion.
My expectation with the above command is, that the pack gets installed into
c:\CMSIS_PACKS\STMicroelectronics\STM32U5xx_DFP\0.1.1
If that folder already exists, then the version is already installed.

On local packs (packs which are not public = in .Web folder but in .Local) we should not have strict versioning expectations, especially if the <url> tag is not set or cannot be reached. This pdsc may not contain the version in the release history.
In this case a local pack file is referenced, therefore we do not need to find out where to download it from. 
If the version installed is higher than what was previously installed you should replace the pdsc file in .Local with the pdsc from the most recently installed pack (to be discussed).

fred-r, 2022-04-05:
Indeed:
- initially, I had a local pack for this U5 DFP: 0.1.0-alpha1
- then I wanted to install a .pack with a higher version (0.1.1)

cpackget rejected the installation saying the version does not exist in my pdsc.

To make it work, I removed the reference to the local pack in "local_repository.pidx".

So, I think that cpackget parsed the pdsc from the .Local instead of parsing the pdsc from the .pack I was trying to install.

jkrech, 2022-04-05:
@fred-r  I actually missed the fact that you referred to a "local_repository.pidx" when you wrote "Local Pack".
It does not change the fact that cpackget should always install a pack which is referenced from the file system.
Even if the same version (or other versions) are listed in the local_repository.pidx. If the same pack version is installed and also available as a local_repository entry, the local_repository entry will take precedence and the installed pack version being ignored.

chaws, 2022-04-05:
Hi @fred-r, @jkrech I confirmed that in the code.

Cpackget hasn't yet extracted the current pack's pdsc file when checking if the version is present in an existing pdsc file.

Let me fix that

jkrech, 2022-04-05:
Hi @chaws , I don't see the purpose of checking for the version you want to install needing to be in an existing pdsc file?
What am I missing? Are you looking for an inconcistency between the version the filename is suggesting and the contained pdsc file you are going to extract into the installation folder?

chaws, 2022-04-05:
@jkrech this is a sanity check/assertion: the version you're trying to install via pack file **should** be present in the pack's pdsc file. The issue here is that cpackget is using an old pdsc file for the validation.

fred-r, 2022-04-06:
> cpackget.exe pack add C:/Packs/STMicroelectronics.STM32U5xx_Drivers.0.1.2.pack
I: Using pack root: "C:\CMSIS_PACKS"
E: The pack's pdsc (C:\CMSIS_PACKS\.Local\STMicroelectronics.STM32U5xx_Drivers.pdsc) has no release tag matching version "0.1.2"
E: pack version not found in the pdsc file

In fact I really need to clean the .Local, updating local_repository.pidx did not work finally.

closed 2022-04-08 by jkrech