Open-CMSIS-Pack/cpackget #51: Following DEVELOPING.md results in invalid executable on macOS

Patater, 2022-04-11:
**Describe the bug**
When following the instructions in `DEVELOPING.md` on macOS, step 6, "Make sure it builds `make build/cpackget`" results in an ELF executable. Executables on macOS should be Mach-O.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone a fresh copy of cpackget.git
2. `make config`
3. `make test-all`
4. `make build/cpackget` 
5. Try running the built cpackget: `./build/cpackget` and see the following error

```
> ./build/cpackget 
exec: Failed to execute process: './build/cpackget' the file could not be run by the operating system.
```

6. Verify the file type with `file ./build/cpackget`.

```
./build/cpackget: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), statically linked, Go BuildID=Zh5jYUyruxOUJMyxI5HE/abJ2yU7uhLeJYLPr5KJ1/4O-QgpEeKIapoQvea0Qq/s1syd2_Gpiva6vPioiUc, not stripped
```

**Expected behavior**
I would expect following the directions in `DEVELOPING.md` to result in a successful build of `cpackget` that I can run on my host system. To accomplish this, `make build/cpackget` could default to building a binary that runs on the host. Alternatively, we could instruct the user to run a platform-specific command, like `make OS=darwin ARCH=amd64 build/cpackget` for macOS amd64.

**Environment (please complete the following information):**
 - OS: darwin
 - Version e1c32e96de41 ("`Make sure pack version exists in the pdsc file and it is the latest one (#46)`")

**Additional context**
After running `make OS=darwin ARCH=amd64 build/cpackget`, we verify the type of the build file with `./build/cpackget` and get `./build/cpackget: Mach-O 64-bit executable x86_64`, which is executable on macOS.

chaws, 2022-04-11:
Hi @Patater thanks for reporting the issue!

OS detection isn't working automatically for MacOS is seems. Could you please try building it with `make OS=darwin  ARCH=amd64 build/cpackget`?

I tried this locally and got it cross-compiled alright:
```bash
$ make OS=darwin  ARCH=amd64 build/cpackget 
Building project
GOOS=darwin GOARCH=amd64 go build -ldflags "-X main.version=`git describe 2>/dev/null || echo unknown`" -o build/cpackget ./cmd/

$ file build/cpackget 
build/cpackget: Mach-O 64-bit x86_64 executable
```

Patater, 2022-04-12:
> @Patater Yes, I think fixing auto-correction would be the way to go here.

OK, I've raised https://github.com/Open-CMSIS-Pack/cpackget/pull/55 to fix this.

chaws, 2022-04-12:
Thanks @Patater for pushing the fix for this.

PR #55 (2022-04-12): make: Default to building for the host
If $OS is not set when running make, the Makefile currently defaults to
building for "linux", not the current host OS. Call out to uname to
determine the OS to use instead, so we can pick a default value for OS
based on the host system. This fixes auto-detection of the OS on macOS.

On Windows, %OS% is already set as an environment variable and we'll
skip calling uname, which is likely not available on Windows.

Fixes #51
commit 2fcae2ac (2022-04-12): make: Default to building for the host

If $OS is not set when running make, the Makefile currently defaults to
building for "linux", not the current host OS. Call out to uname to
determine the OS to use instead, so we can pick a default value for OS
based on the host system. This fixes auto-detection of the OS on macOS.

On Windows, %OS% is already set as an environment variable and we'll
skip calling uname, which is likely not available on Windows.

Fixes #51