Open-CMSIS-Pack/cpackget #56: cpackget index -f <url> does not work on Windows

JonatanAntoni, 2022-04-12:
**Describe the bug**
Running `cpackget` on Windows to update the `index.pidx` file fails with:
```bash
$ cpackget index -f https://www.keil.com/pack/index.pidx
I: Using pack root: "C:\Users\xxx\AppData\Local\Arm\Packs"
I: Updating index [https://www.keil.com/pack/index.pidx]
I: Overwriting public index file C:\Users\xxx\AppData\Local\Arm\Packs\.Web\index.pidx
E: Can't move file "C:\Users\xxx\AppData\Local\Arm\Packs\.Download\index.pidx" to "C:\Users\xxx\AppData\Local\Arm\Packs\.Web\index.pidx": rename C:\Users\xxx\AppData\Local\Arm\Packs\.Download\index.pidx C:\Users\xxx\AppData\Local\Arm\Packs\.Web\index.p
idx: Access is denied.
E: rename C:\Users\xxx\AppData\Local\Arm\Packs\.Download\index.pidx C:\Users\xxx\AppData\Local\Arm\Packs\.Web\index.pidx: Access is denied.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run the cpackget command
2. See error

**Expected behavior**
It is expected that `cpackget` overwrites the existing `index.pidx`.

**Environment (please complete the following information):**
 - OS: Windows
 - Version 20H2 (19042.1586)

jkrech, 2022-04-13:
@chaws, please note that you need to manage file permissions in the $CMSIS_PACK_ROOT folder. My bad, I guess. Since we don't want to people temper with the content of the pack folder, we set file permissions to read-only. However in case files/directories need to be modified or deleted, the read-only flag needs to be removed or the operation forced. Note that cpackget shall also set read-only flags on downloaded and installed files and directories.
In this case the user has probably used a different tool (MDK pack installer) to setup the pack directory and therefore sees the permission problem.

JonatanAntoni, 2022-04-13:
@chaws, @jkrech, indeed, the pack folder was initially created by MDK's PackInstaller. I'll update the bug report above, accordingly.

Sounds like we should somehow pin down the spec (layout) of the pack folder including such tweaks like setting files read-only. This should not only be done on Windows but on all platforms to prevent a user from unintentionally modifying shared files.

JonatanAntoni, 2022-04-13:
I confirm, removing the `read-only` file flag manually solves the issue.

jkrech, 2022-05-20:
@chaws  do you have a timeline for addressing this defect?

chaws, 2022-06-09:
@jkrech This took a bit more effort than I imagined, but I'm close to send a PR with setting the read-only flag in the CMSIS_PACK_ROOT installation folder.

Before I push the PR, I wanted to check if there will be any file that won't need to have read-only flag set?
I'm changing cpackget to set read-only for **all files and directories** including `.Web/*`, `.Local/*`, `pack.idx`, `.Download/*` and the installed packs, including all sub-directories.

jkrech, 2022-06-09:
@chaws this sounds reasonable, as we don't want people to manually edit the content of the CMSIS-Pack root folder. Being primarily a windows user, I maybe overlooking a specific use-case, since the windows permission handling is not great and read-only is ignored by many commands e.g. `del`

closed 2022-07-26 by lud0v1c