Open-CMSIS-Pack/cpackget #57: cpackget cannot install older pack versions

JonatanAntoni, 2022-04-13:
**Describe the bug**
`cpackget pack add` cannot download a pack version older than the latest version listed in the `index.pidx`.

The error message contains a `cpackget` command line that is not functional:
`cpackget --force index https://keil.com/pack/index.pidx`

**To Reproduce**
Steps to reproduce the behavior:
1. Initialize a new pack repo: `cpackget -R .packs init https://keil.com/pack/index.pidx`
2. Install some older pack version: `cpackget -R .packs pack add ARM.V2M_MPS3_SSE_300_BSP.1.2.0` (note, at this time version 1.3.0 is the latest version of that pack)
3. See error message:
   ```
   I: Using pack root: ".packs"
   I: Adding pack "ARM.V2M_MPS3_SSE_300_BSP.1.2.0"
   E: URL for the pack cannot be determined. Please consider updating the public index. Ex: cpackget --force index 
   https://keil.com/pack/index.pidx
   ```
4. Install the latest version of the pack: ` cpackget -R .packs pack add ARM.V2M_MPS3_SSE_300_BSP`
   ```
   I: Using pack root: ".packs"
   I: Downloading ARM.V2M_MPS3_SSE_300_BSP.pdsc 100% || (25/25 kB, 4.686 MB/s)
   I: Adding pack "ARM.V2M_MPS3_SSE_300_BSP.1.3.0"
   I: Downloading ARM.V2M_MPS3_SSE_300_BSP.1.3.0.pack 100% || (4.2/4.2 MB, 24.731 MB/s)
   I: Extracting files to .packs\ARM\V2M_MPS3_SSE_300_BSP\1.3.0 100% || (114/114, 614 it/s)
   ```
5. Install the older pack version again: `cpackget -R .packs pack add ARM.V2M_MPS3_SSE_300_BSP.1.2.0`
   ```
I: Using pack root: ".packs"
I: Adding pack "ARM.V2M_MPS3_SSE_300_BSP.1.2.0"
I: Downloading ARM.V2M_MPS3_SSE_300_BSP.1.2.0.pack 100% || (5.3/5.3 MB, 24.428 MB/s)
I: Extracting files to .packs\ARM\V2M_MPS3_SSE_300_BSP\1.2.0 100% || (102/102, 538 it/s)
   ```

**Expected behavior**
`cpackget` is expected to download/install *all* ever published versions of a pack right into an empty pack folder. Regardless of the version requested, `cpackget` shall download/update the packs `.PDSC` file in `.Web` folder.

The error message is expected to give a proper command: 
`cpackget index --force https://keil.com/pack/index.pidx`

The `--force` flag on `index` seems to be always required. Right after `init` there is an empty `index.pidx` so a user would never be able to update the index without giving the `-f` flag. It is expected that this is default behaviour so `-f` flag can be removed.

**Environment (please complete the following information):**
 - OS: Windows
 - Version 2020H2

**Additional context**
Once the pack's `.PDSC` file is present in the `.Web` folder the installation works as expected.

chaws, 2022-04-13:
Thanks for raising the issue, cpackget was blindly looking for the version in index.pidx specified during installation. I have opened a PR #60 addressing this issue

closed 2022-04-13 by chaws

chaws, 2022-04-13:
@jkrech you are correct and this is how it has been implemented. I accidentally introduced a bug that is triggered when cpackget installs a pack with packID version older than the one specified in the PDSC tag in index.pidx and there is not pack PDSC file under .Web. cpackget searches for the PDSC tag in `.Web/index.pidx` in order to find the URL to download the PDSC file. I'll illustrate the bug with the example Jonatan presented in the issue:

1. Init the pack-root folder: `cpackget -R .packs init https://keil.com/pack/index.pidx`
2. The PDSC tag is present: `<pdsc url="http://www.keil.com/pack/" vendor="ARM" name="V2M_MPS3_SSE_300_BSP" version="1.3.0" />`
3. With an empty pack-root folder, install `cpackget -R .packs pack add ARM.V2M_MPS3_SSE_300_BSP.1.2.0`
4. Note that **1.2.0** is being installed, but the tag in index.pidx contains version **1.3.0**
5. cpackget wrongly tries to find the PDSC tag with version **1.2.0** `<pdsc url="http://www.keil.com/pack/" vendor="ARM" name="V2M_MPS3_SSE_300_BSP" version="1.2.0" />`, but it doesn't find it so it can't retrieve `ARM.V2M_MPS3_SSE_300_BSP.pdsc`

Bottom line, cpackget was using pack's _vendor_, _name_ and _version_ to search for the pdsc tag in index.pidx, where it should actually use _vendor_ and _name_ only. That's what the PR is fixing.

My tests didn't catch this because they always install the latest version of packs therefore it always worked. It is a tricky bug.

jkrech, 2022-04-14:
Thanks a lot. Now I understand and I feel relieved that we do not have a conceptual misunderstanding.