Open-CMSIS-Pack/cpackget #58: cpackget returns with error state on already installed packs

JonatanAntoni, 2022-04-13:
**Describe the bug**
`cpackget` returns with error state when asked to install a pack that is already installed. This leads to CI pipelines to fail.

**To Reproduce**
Steps to reproduce the behavior:
1. `cpackget -R .packs init https://keil.com/pack/index.pidx`
2. `cpackget -R .packs pack add ARM.CMSIS` install the pack:
   ```
   I: Using pack root: ".packs"
   I: Downloading ARM.CMSIS.pdsc 100% |█████████████████████████████████| (238/238 kB, 338.772 kB/s)
   I: Adding pack "ARM.CMSIS.5.8.0"
   I: Downloading ARM.CMSIS.5.8.0.pack 100% |██████████████████████████████| (34/34 MB, 30.838 MB/s)
   I: Extracting files to .packs\ARM\CMSIS\5.8.0 100% |█████████████████████| (8133/8133, 207 it/s)
   ```
4. `cpackget -R .packs pack add ARM.CMSIS` recognizes pack is already installed
5. See error:
   ```
   I: Using pack root: ".packs"
   I: Adding pack "ARM.CMSIS.5.8.0"
   E: pack ARM.CMSIS.5.8.0 is already installed here: .packs\ARM\CMSIS\5.8.0
   ```

**Expected behavior**
Reinstalling an already installed pack should not be treated as an error. In the first place it may be a no-op with just info (or at max warning) message. In the second place their might be situations where a user wants to force reinstall a pack. While this can already be achieved by removing and install the pack in two steps, a `--force` flag for the `add` command may be convenient. Today, the `add -f` flag conflicts with the usual abbreviation for `--force`.

**Environment (please complete the following information):**
 - OS: Windows
 - Version: 2020H2

**Additional context**
Add any other context about the problem here.

chaws, 2022-04-13:
Thanks for raising this @JonatanAntoni ! 
I have changed it so that the error message becomes an info message e.g. `` the PR #59 addresses that.

I'll address the second part of your issue on another PR. What do you think of `--force-reinstall`? pip[1] uses that option for exactly the same purpose as you described 

[1] https://pip.pypa.io/en/stable/cli/pip_install/?highlight=reinstall#cmdoption-force-reinstall

JonatanAntoni, 2022-04-13:
Yes, doing this similar to other package managers seems like a good idea.

closed 2022-05-11 by chaws