Open-CMSIS-Pack/cpackget #63: Allow filtering when listing packs

chaws, 2022-04-27:
Extracted from https://github.com/Open-CMSIS-Pack/cpackget/issues/35

>cpackget list arm show all packs from Arm
cpackget list *st* show all packs that include the letters st in the file name

jkrech, 2022-04-29:
a) We need to define whether we want to search case sensitive or not
b) We should consider consistency between tools.
```
csolution list packs -f/--filter arg
```

chaws, 2022-04-29:
@brondani could you please point me to the implementation of `csolution list packs -f arg`? I'd like to apply the same pattern-matching to cpackget.

brondani, 2022-04-29:
> @brondani could you please point me to the implementation of `csolution list packs -f arg`? I'd like to apply the same pattern-matching to cpackget.

@chaws Here you can find the filter implementation:
https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/src/ProjMgrWorker.cpp#L1319-L1343
It checks whether all filter words (independently of the order) are present in the original list of strings. It's case sensitive.

lud0v1c, 2022-05-24:
@jkrech if we want to mirror csolution's behaviour, should `cpackget list -f`:
- Accept several words
- Be case sensitive
- Maybe a wildcard (*) would be nice, to facilitate searching for similar packs

jkrech, 2022-05-24:
I suggest you talk to @brondani to align the implementation/features supported by cpackget and csolution.
See the documentation here:
https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/Overview.md#commands

lud0v1c, 2022-05-24:
Me and @chaws discussed it a bit after I left my comment, and I agree with you that wildcards are not needed, as the behavior you described is pretty much it. I will reach out to him to confirm.

chaws, 2022-05-27:
Addressed by https://github.com/Open-CMSIS-Pack/cpackget/pull/92

closed 2022-05-27 by chaws