Open-CMSIS-Pack/cpackget #67: Same local pack added several times

jeromecoutant, 2022-05-03:
Hi

Working with local packs:

`cpackget pdsc add <pdsc file>`

We got 
`<pdsc vendor="STMicroelectronics" url=file://localhost/xxx" name="xxx" version="0.0.1"></pdsc>`
in the local_repository.pidx 

We update the pdsc file, to increase version for example.
We need then to update the version in our local_repository.pidx 

`cpackget pdsc add <same pdsc file>`

We got the same file 2 times:
```
<pdsc vendor="STMicroelectronics" url=file://localhost/xxx" name="xxx" version="0.0.1"></pdsc>
<pdsc vendor="STMicroelectronics" url=file://localhost/xxx" name="xxx" version="0.0.2"></pdsc>
```

@VVESTM

jkrech, 2022-05-03:
Thanks for reporting. I had discussed this use case with @chaws end of last week and I do expect this to be fixed soon.
The basic rule is that you cannot register the same path + <vendor>.<packname>.pdsc  twice.
At the same time we have also discussed that the version attribute in the local_repository.pidx is "meaningless". 
While packs installed in $CMSIS_PACK_ROOT are read-only the purpose of the local_repository.pidx is to reference 'rw' packs. Therefore all tools shall fully rely on the latest version described in the pdsc file which may be different from the time of adding the pdsc file.

Please let me know in case you have any comments or questions.

chaws, 2022-05-05:
This issue was fixed in https://github.com/Open-CMSIS-Pack/cpackget/pull/74

```bash
# List 1.2.4 installed
$ cpackget list
I: Using pack root: "../packroot"
I: Listing installed packs
I: TheVendor::DevPack@1.2.4 (installed via /home/chaws/linaro/src/arm/cpackget/testdata/devpack/1.2.4/TheVendor.DevPack.pdsc)

# Copy the PDSC file to a new location so we can edit it
$ cp testdata/devpack/1.2.4/TheVendor.DevPack.pdsc /tmp/

# Update the version to 1.2.5
$ sed -i 's/1.2.4/1.2.5/g' /tmp/TheVendor.DevPack.pdsc

# Installing should be OK
$ cpackget add /tmp/TheVendor.DevPack.pdsc 
I: Using pack root: "../packroot"
I: Adding pdsc "/tmp/TheVendor.DevPack.pdsc"

# Make sure it is installed
$ cpackget list
I: Using pack root: "../packroot"
I: Listing installed packs
I: TheVendor::DevPack@1.2.4 (installed via /home/chaws/linaro/src/arm/cpackget/testdata/devpack/1.2.4/TheVendor.DevPack.pdsc)
I: TheVendor::DevPack@1.2.5 (installed via /tmp/TheVendor.DevPack.pdsc)

# Now update it to 1.2.6
$ sed -i 's/1.2.5/1.2.6/g' /tmp/TheVendor.DevPack.pdsc 

# Installing should be prevented this time
$ cpackget add /tmp/TheVendor.DevPack.pdsc 
I: Using pack root: "../packroot"
I: Adding pdsc "/tmp/TheVendor.DevPack.pdsc"
I: pdsc already in index

# Make sure it did not get installed, but confirm that the version got updated to 1.2.6
$ cpackget list
I: Using pack root: "../packroot"
I: Listing installed packs
I: TheVendor::DevPack@1.2.4 (installed via /home/chaws/linaro/src/arm/cpackget/testdata/devpack/1.2.4/TheVendor.DevPack.pdsc)
I: TheVendor::DevPack@1.2.6 (installed via /tmp/TheVendor.DevPack.pdsc)
```