Open-CMSIS-Pack/cpackget #68: `cpackget` not working with corporate VPN

jankap, 2022-05-03:
`cpackget` seems to be unable to download packages when called from behind a corporate VPN (Cisco AnyConnect) + Proxy (Zscaler):

VPN enabled:
```
PS D:\git\test-project\prj> cpackget pack add Keil.STM32F1xx_DFP.2.4.0
I: Using pack root: "C:\Users\<user>\AppData\Local\Arm\Packs"
I: Adding pack "Keil.STM32F1xx_DFP.2.4.0"
E: Get "http://www.keil.com/pack/Keil.STM32F1xx_DFP.2.4.0.pack": dial tcp 23.11.231.137:80: connectex: No connection could be made because the target machine actively refused it.
E: failed to download file
```

Now, VPN disabled:
```
PS D:\git\test-project\prj> cpackget pack add Keil.STM32F1xx_DFP.2.4.0
I: Using pack root: "C:\Users\<user>\AppData\Local\Arm\Packs"
I: Adding pack "Keil.STM32F1xx_DFP.2.4.0"
E: Can't decompress "C:\Users\<user>\AppData\Local\Arm\Packs\.Download\Keil.STM32F1xx_DFP.2.4.0.pack": zip: not a valid zip file
E: fail to decompress file
```

Keil.STM32F1xx_DFP.2.4.0.pack is now 0 byte in size. Seems like it does not get re-downloaded or size checked.

VPN disabled + deleting Keil.STM32F1xx_DFP.2.4.0.pack:
```
PS D:\git\test-project\prj> cpackget pack add Keil.STM32F1xx_DFP.2.4.0
I: Using pack root: "C:\Users\<user>\AppData\Local\Arm\Packs"
I: Adding pack "Keil.STM32F1xx_DFP.2.4.0"
I: Downloading Keil.STM32F1xx_DFP.2.4.0.pack 100% |████████████████████████████████████| (48/48 MB, 7.387 MB/s)
I: Extracting files to C:\Users\<user>\AppData\Local\Arm\Packs\Keil\STM32F1xx_DFP\2.4.0 
```

Works as expected. So I guess there are two issues:
1. System proxy settings are not used 
2. Erroneous downloads are not overwritten

cpackget version 0.4.0
Windows 10 21H2

Thanks for this great initiative!

chaws, 2022-05-03:
Hi @jankap,

> 1. System proxy settings are not used

How are you configuring your proxy? I have written down a few notes on it: https://github.com/Open-CMSIS-Pack/cpackget#working-behind-a-proxy

> 2. Erroneous downloads are not overwritten

Good catch! cpackget only removes the downloaded file if it fails to unzip it. I'll fix this to remove on http errors as well

Thanks for raising the issue!

jankap, 2022-05-03:
> Hi @jankap,
> 
> > 1. System proxy settings are not used
> 
> How are you configuring your proxy? I have written down a few notes on it: https://github.com/Open-CMSIS-Pack/cpackget#working-behind-a-proxy

I have to admit that I did overlook that section. Nevertheless, `$http_proxy` should be defined. Let me check.

> 
> > 2. Erroneous downloads are not overwritten
> 
> Good catch! cpackget only removes the downloaded file if it fails to unzip it. I'll fix this to remove on http errors as well
> 
> Thanks for raising the issue!

You're welcome :) Glad I can give a little bit back.

chaws, 2022-05-03:
Hi @jankap I already fixed erroneus downloaded file here: https://github.com/Open-CMSIS-Pack/cpackget/pull/69. It should be rolled out on next cpackget release.

With regard to `$http_proxy`, were you able to confirm it's available to `cpackget`?