Open-CMSIS-Pack/cpackget #70: Remove local packs using PDSC file name

chaws, 2022-05-03:
Currently users would add local packs using 
```bash
$ cpackget pdsc add path/to/ARM.CMSIS.pdsc
```
and then remove it by running 
```bash
$ cpackget pdsc rm ARM.CMSIS
```
Thus not specifying the PDSC path at all.
Since it's possible that the same path can be added multiple times using different paths,
```bash
$ cpackget pdsc add one/path/to/ARM.CMSIS.pdsc
$ cpackget pdsc add another/path/to/ARM.CMSIS.pdsc
```
There is no way to specify which `ARM.CMSIS.pdsc` is intended for removal.

cpackget should then:
1. Allow users to remove local packs using PDSC file path notation: `$ cpackget pdsc rm ARM.CMSIS.pdsc`
2. And if there's such case where a single pack was installed using more than one PDSC file path, then cpackget should require the full path of that PDSC file in order to remove it
```bash
$ cpackget pdsc rm /absolute/path/to/ARM.CMSIS.pdsc
```

The user can retrieve the absolute path of the local pack by listing all installed packs (to be implemented by https://github.com/Open-CMSIS-Pack/cpackget/pull/64#issuecomment-1116484640)

jkrech, 2022-05-04:
@chaws are we in agreement that once #64 is committed, the 'pdsc' command would vanish from the above?
```
$ cpackget rm /absolute/path/to/ARM.CMSIS.pdsc
$ cpackget rm ARM.CMSIS.pdsc
```
For consistency we need to check that both cpackget 'add' and 'rm' work in a consistent way. When removing a pack with an "imprecise" identifier where multiple installed/registered packs match, does cpackget remove all of them or requires the user to precisely specify the pack/pdsc?

If `cpackget rm ARM.CMSIS` or `cpackget rm ARM::CMSIS` removes all installed pack versions of the ARM.CMSIS pack then
`cpackget rm ARM.CMSIS.pdsc` should also remove all file references an ARM.CMSIS.pdsc file in the local_repository.pidx

chaws, 2022-05-04:
Hi @jkrech 

> are we in agreement that once https://github.com/Open-CMSIS-Pack/cpackget/pull/64 is committed, the 'pdsc' command would vanish from the above?

Yes! Both `pdsc` and `pack` subcommands will be in "deprecated" state. They aren't going to be displayed anywhere in cpackget's help command/usage but you'll still be able to invoke them, this is until we decide we can remove them completely.

> does cpackget remove all of them or requires the user to precisely specify the pack/pdsc?

For `cpackget rm ARM.CMSIS` would remove any installed pack within `CMSIS_PACK_ROOT/ARM/CMSIS/*`, then I agree that `cpackget rm ARM.CMSIS.pdsc` is invoked, this would remove any reference of ARM.CMSIS from the local_repository.pidx.

jkrech, 2022-05-04:
Agreed then.

closed 2022-05-05 by chaws