Open-CMSIS-Pack/cpackget #79: Support multiple package indexes

chaws, 2022-05-11:
This issue was asked by @jeromecoutant.

Some vendors might have their own package index. Ex: Let's name https://www.keil.com/pack/index.pidx as Keil.pidx. It contains all known public packs. But ST might have their own package index, let's say ST.pidx.

While Keil.pidx contains ARM.CMSIS and hundreds of other packs, it might not contain an ST-only pack e.g. STM.Pack. Whereas ST.pidx contains STM.Pack, it won't list ARM.CMSIS in it.

chaws, 2022-05-11:
@jkrech after you suggestion of using vidx2pidx to merge those 2 into a single index, it got me thinking about updating a list of all packs. Given that the merged pidx will only contain a single URL tag, which URL should that point to?

jkrech, 2022-05-11:
All vendors must maintain vendor.pidx files that only list packs from one vendor.
A maintainer of an index.pidx file need to maintain a provider.vidx file 
The expectation is that both the provider.vidx and the resulting index.pidx are hosted at the same <url>
Therefore the answer to your question is, that the  <url> tag in the index.pidx is inherited from the vidx file that was used to generate the index.pidx.
I hope this makes sense?
Any consumer of an index.pidx file needs to be able to know which vendor's pidx files have been used to compile it.

jeromecoutant, 2022-05-12:
Hi
My first thought was that, in CMSIS_PACK_ROOT, in .Web, instead of having only 1 index.pidx, there is Keil.pidx, STMicroelectronics.pidx, ARM.pidx, ....
Then when user is using `cpackget pack add <vendor>.<name>`, tool can get information in the correct vendor pidx file.

jkrech, 2022-05-12:
Hi Jerome,
I did understand your proposal but looking at it from an end user's perspective I don't want them being exposed to this. We should not present "fragmentation". At the same time we need to be transparent. I hope this is an acceptable compromise.

jkrech, 2022-05-12:
Please note if a tool does not find a valid <url> tag in the index.pidx I think there is simply no way for the tool to execute an update and should report this as an error.

jkrech, 2022-05-12:
@jeromecoutant and @chaws  are you happy to close this issue or specify what actions need to be taken?
e.g. @chaws do you need to raise an issue against vidx2pidx to derive the <url> tag content from the input vidx file? Which may also be an empty string.

chaws, 2022-05-18:
@jkrech sorry for late answer on this issue.

But while I was building up an example for Jerome in order to merge such `Keil.pidx` and `ST.pidx` into a single `index.pidx` file using vidx2pidx I found a problem: the content of Keil.pidx (https://www.keil.com/pack/Keil.pidx) contains much smaller number of packs if compared to index.pidx (https://www.keil.com/pack/index.pidx). Is that expected? I can't use the latter in a vidx file because vidx2pidx uses the [content of the Vendor tag in vidx file to build the URL of the pidx file](https://github.com/Open-CMSIS-Pack/vidx2pidx/blob/main/cmd/pidx.go#L108).

So I think the solution here would be adding an ST tag such as:
```xml
<pidx vendor="ST" url="path/to/index/" />
```
Into `<vindex>` of https://www.keil.com/pack/Keil.vidx and then use vidx2pidx to generate a flat index.pidx file with ST packs int it as well. It would be ST's responsibility to maintain the file up-to-date against the original one in Keil.vidx. 

If [Keil.pidx](https://www.keil.com/pack/Keil.pidx) contained the same contents of https://www.keil.com/pack/index.pidx, then ST would simply maintain their own vidx file pointing to Keil.pidx.

chaws, 2022-06-07:
I'm closing down this issue. For now it will not be possible to have multiple indexes in cpackget.

closed 2022-06-07 by chaws