Open-CMSIS-Pack/cpackget #83: Investigate why pack extraction is so much slower on Windows

chaws, 2022-05-17:
Pack installation on Linux is much faster if compared to Windows. Part of the installation process consists of extracting the files in the pack, and this is done via Golang's stdlib. Perhaps that is not the best way to do on Windows?

For example, when writing files on Windows it spawns a new process for that, so if the cpackget tries to change the same file's content, it fails on Windows but not on Linux.

jkrech, 2022-05-31:
My assumption it is because the virus protection does not trust unsigned executables.

chaws, 2022-06-07:
I think you could first attempt to use a native unzip utility already available on Windows.

lud0v1c, 2022-06-08:
Tested adding different sized packs and some Windows native ways to unzip:
* `powershell.exe -nologo -noprofile -command "& { Add-Type -A 'System.IO.Compression.FileSystem'; [IO.Compression.ZipFile]::ExtractToDirectory(p.path, packHomeDir); }"` -> Comes with Windows 8 and later
* `tar.exe -xf` -> Comes with Windows 10 build 17063 and later
* `Expand-Archive -Path p.path -DestinationPath packHomeDir` -> Also Windows 10+

Overall, all of these methods are faster than the current stdlib method, however by a very small margin on larger packs (~10%) and the smallest packs varied around 20-30% (from 4.25 seconds to 3.19, for example).

@jkrech @chaws I can change the unzip method when it runs on Windows to the first one, so it's compatible with any Windows 8 and later. It will give a performance boost, with the "tradeoff" of not supporting very old, unsupported Windows versions.

jkrech, 2022-06-09:
a) I am not really too concerned about performance (yet?)
b) I do not think that we have to support any very old/unsupported Windows versions, however let's make sure we communicate/document this clearly. I don't think we currently detail OS versions being supported by the tools.

lud0v1c, 2022-06-15:
Closing for now until further need to revamp the extraction process.

closed 2022-06-15 by lud0v1c