Open-CMSIS-Pack/cpackget #85: Cannot scroll up after displaying license and error message if I declined

jkrech, 2022-05-20:
Windows cmd:
`cpackget add Arm-Packs::Unity`
Pressing key 'd'
```
E: user does not agree with the pack's license
E: user does not agree with the pack's license

C:\Open-CMSIS-Pack\ctools\bin>
```
I cannot scroll further up. The first line is the E: which also gets duplicated
Note I am not sure I would consider this an error, as the operation selected by the user was executed successfully.

chaws, 2022-05-20:
I tested on Linux and scrolling seems fine after declining the license. Let me report back after replicating this issue on Window.

lud0v1c, 2022-06-02:
Tried this out with Powershell and it happens as described, after declining the error message gets duplicated and the current screen gets wiped.. Investigating why

chaws, 2022-06-09:
I have done some experimenting and we the only workaround I could work out is opening a new `cmd` window just with the license. Then after user's response, the window would close out and cpackget would continue from there.

Does that seem reasonable to you @jkrech ?

jkrech, 2022-06-09:
@chaws I am not sure how this could work if I am in a terminal session to the machine? Where would that other window pop up? I think you would need to run the cmd in the same window?

chaws, 2022-06-09:
@jkrech this would only be applied if you're running a GUI. The idea is:
1. Window 1 (git-bash or cmd): `cpackget add <pack-with-license>.pack`
2. Window 2 will be a new cmd terminal, displaying only the license window like the image below. Meanwhile, cpackget on Window 1 will be waiting for user input on Window 2
3. User accepts/declines/extracts the license
4. Window 2 close out
5. Window 1 continue with installation/extraction. Afterwards, scrolling will not be affected.

[image]

chaws, 2022-06-09:
I talked to Luis, we both did some research on why scrolling gets messy, but we couldn't find the reason why unfortunately. If I was to guess, it is due to some hidden configuration on the terminal. Note that this does not happen in git-bash.

jkrech, 2022-06-09:
Assuming we can rely on the detection of "running with GUI", this makes sense then. But it does not change the problem for the none-GUI mode then, right?

chaws, 2022-06-09:
The [interactive-mode detection](https://github.com/Open-CMSIS-Pack/cpackget/blob/main/cmd/utils/utils.go#L306) on cpackget is what it seems the common practice. While it has a false positive on git-bash (hence https://github.com/Open-CMSIS-Pack/cpackget/issues/95), it shouldn't try to pop up a new window on CI, for instance.

This behavior will only be applied on Windows platforms.

lud0v1c, 2022-06-22:
Error message is now a normal info message ([eaa7f7e](https://github.com/lud0v1c/cpackget/commit/eaa7f7e12419d9778b0f4ecf3cf7442897e17286)).