Open-CMSIS-Pack/cpackget #89: cpackget doesn't download the latest pack version

jeromecoutant, 2022-05-24:
Hi

In my CMSIS_PACK_ROOT, I have this pack installed:
```
|-- STMicroelectronics
|   `-- basic_trace
|       `-- 0.0.1
```
My index.pidx is correct I think:
```
$ grep basic_trace .Web/index.pidx
        <pdsc url="https://xxx/STMicroelectronics.basic_trace/" vendor="STMicroelectronics" name="basic_trace" version="0.1.0"/>
```

I think cpackget should install the new version, but it doesn't:
```
$ cpackget.exe -V
cpackget version 0.4.1

$ cpackget.exe pack add STMicroelectronics.basic_trace -v
D: Reading pidx from file "C:\xxx\.Web\index.pidx"
D: Registring "STMicroelectronics.basic_trace.0.1.0"
D: Specified packs [STMicroelectronics.basic_trace]
D: Extracting pack info from "STMicroelectronics.basic_trace"
D: Listing files and directories in "C:\xxx\.Web" that match "^.*\.pdsc$"
D: Found "STMicroelectronics.basic_trace.pdsc" in "C:\xxx\.Web"
D: Checking if STMicroelectronics.basic_trace. is installed
I: Adding pack "STMicroelectronics.basic_trace"
I: pack STMicroelectronics.basic_trace is already installed here: C:\xxx\STMicroelectronics\basic_trace
```

I would expect something like:
```
D: Found "STMicroelectronics.basic_trace.pdsc" in "C:\xxx\.Web"
D: Checking if STMicroelectronics.basic_trace.0.1.0 is installed
I: Adding pack "STMicroelectronics.basic_trace.0.1.0"
```

chaws, 2022-05-24:
Hi @jeromecoutant could you please retry it by running

```bash
$ cpackget add STMicroelectronics.basic_trace@latest
```

From what I remember in previous discussions with @jkrech , if you don't specify the version then any installed version already satisfies the installer.

jeromecoutant, 2022-05-24:
````
D: Registring "STMicroelectronics.basic_trace.0.1.0"
D: Specified packs [STMicroelectronics.basic_trace@latest]
D: Extracting pack info from "STMicroelectronics.basic_trace@latest"
E: bad pack name: it should be one of the following formats: Vendor.Pack, Vendor.Pack.x.y.z, Vendor.Pack.x.y.z.pack, or 
Vendor.Pack.pdsc
````

> if you don't specify the version then any installed version already satisfies the installer.

https://github.com/Open-CMSIS-Pack/cpackget/blob/main/README.md?plain=1#L77
=> Install the latest published version of a public package listed in the package index:

jkrech, 2022-05-24:
@chaws looks like you  first compute what "latest" resolves to (0.1.0) but then you keep that info and try to extract a pack that has the version "latest", which is not a valid version format.

chaws, 2022-05-25:
Argh, sorry about that!

Jerome, the documentation is outdated. Please @jkrech correct me if I'm wrong, but once we added `@`, `@~`, and `>=` to cpackget, we decided that some behaviors would change. Before this feature, installing a pack with `cpackget pack add ARM.CMSIS` would *always* install the latest one. But after the new features kicked in, that syntax would be satisfied if any `ARM.CMSIS` pack was already installed. And to suppress the use case for adding the latest version, we added `@latest`.

I specified a wrong syntax for you @jeromecoutant ! The correct one should be:
```bash
$ cpackget pack add STMicroelectronics::basic_trace@latest
```

See an example I have tried locally:
```bash
$ cpackget --version
cpackget version v0.4.1

$ cpackget pack list
I: Using pack root: "../packroot"
I: Listing installed packs
I: ARM.CMSIS.4.3.0
I: ARM.CMSIS.4.4.0

$ cpackget pack add ARM::CMSIS
I: Using pack root: "../packroot"
I: Adding pack "ARM::CMSIS"
I: pack ARM::CMSIS is already installed here: ../packroot/ARM/CMSIS

$ cpackget pack add ARM::CMSIS@latest
I: Using pack root: "../packroot"
I: Adding pack "ARM::CMSIS@latest"
I: Extracting files to ../packroot/ARM/CMSIS/5.8.0 100% |█████████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (8133/8133, 9821 it/s)        

$ cpackget pack list
I: Using pack root: "../packroot"
I: Listing installed packs
I: ARM.CMSIS.4.3.0
I: ARM.CMSIS.4.4.0
I: ARM.CMSIS.5.8.0
```

jeromecoutant, 2022-05-25:
> $ cpackget pack add STMicroelectronics::basic_trace@latest

Yes, this command is working

jeromecoutant, 2022-05-25:
More difficult use case :-)

My current tree:
```
|-- STMicroelectronics
|   |-- STM32U5xx_Drivers
|   |   `-- 2.0.0-alpha.0.5
```

My current pidx:
```
$ grep STM32U5xx_Drivers .Web/index.pidx
        <pdsc url="https://xxx/" vendor="STMicroelectronics" name="STM32U5xx_Drivers" version="2.0.0-alpha.0.6"/>
```

```
cpackget.exe pack add STMicroelectronics::STM32U5xx_Drivers@latest -v
D: Reading pidx from file "C:\xxx\.Web\index.pidx"
D: Registring "STMicroelectronics.STM32U5xx_Drivers.2.0.0-alpha.0.6"
D: Specified packs [STMicroelectronics::STM32U5xx_Drivers@latest]
D: Extracting pack info from "STMicroelectronics::STM32U5xx_Drivers@latest"
D: Listing files and directories in "C:\xxx\.Web" that match "^.*\.pdsc$"
D: Found "STMicroelectronics.STM32U5xx_Drivers.pdsc" in "C:\xxx\.Web"
D: Checking if STMicroelectronics.STM32U5xx_Drivers. is installed
D: Listing files and directories in "C:\xxx\STMicroelectronics\STM32U5xx_Drivers" that match ".*"
D: Checking if the latest version is installed
D: Initializing PdscXML object for "C:\xxx\.Web\STMicroelectronics.STM32U5xx_Drivers.pdsc"
D: Reading pdsc from file "C:\xxx\.Web\STMicroelectronics.STM32U5xx_Drivers.pdsc"
I: Adding pack "STMicroelectronics::STM32U5xx_Drivers@latest"
I: pack STMicroelectronics::STM32U5xx_Drivers@latest is already installed here: C:\xxx\STMicroelectronics\STM32U5xx_Drivers\2.0.0-alpha.0.5
```

jeromecoutant, 2022-05-27:
Note I think you don't have to compare 2.0.0-alpha.0.5 with 2.0.0-alpha.0.6,
you just have to trust .Web/index.pidx information

chaws, 2022-05-27:
Could you please past the debug part where it says 

```bash
D: Resolving version modifier for ...
D: - resolved(@latest) as ...
```

?

I think this is because the index.pidx is more up-to-date if compared to the pack pdsc file within `.Web/`

jkrech, 2022-05-30:
I think the challenge here is:
a) for downloading a pack, cpackget must not rely on the information contained in the index.pidx, because a <release> tag may contain an explicit download link that overrides the url attribute from the index.pidx
b) as a consequence cpackget needs to ensure that the pdsc file for a pack is available in the .Web folder to read the <releases> section
c) it looks like cpackget is currently not checking that the version attribute of the <pdsc> tag in the index.pidx matches the most recent version (=  version attribute of the first <release> tag). If that is not the case, cpackget needs to download the pdsc file  (url/vendor.name.pdsc) to retrieve the latest public version of the pdsc.

I think this up-to-date check for the pdsc is currently missing.
If you would remove the pdsc file from the .Web folder cpackget already now will ensure that the pdsc file gets downloaded.

chaws, 2022-05-30:
Yes, the feature is missing indeed. It is not taking into consideration that after index.pidx gets updated, the Vendor.Pack.pdsc might get outdated. So updating it after checking the version in index.pidx should be the best approach here.

jkrech, 2022-05-30:
Note that this is the "sparse" update mode.
The 'update' implemented in other tools like MDK or eclipse, where the UI displays packs, devices, boards and examples from all public packs require to download the latest version of the index.pidx and then check for each pdsc file in the .Web folder whether 
a) it does not exist -> download
b) it is not the latest version -> download
c) it exists, but is not longer listed in the index.pidx -> remove

chaws, 2022-05-30:
Perhaps all the three behaviors could be implemented and executed by default when updating the index?

jkrech, 2022-05-30:
Executing a `full` update versus a `sparse` check for updates shall be an option for the user.
Assuming a CI system running a full check for updates seems like a waste of time and resource.

jkrech, 2022-05-31:
@jeromecoutant thanks for following up. I believe in the meantime we had figured out the missing pdsc update step, which you confirmed with your manual step. Even though in your case just taking the information form the index.pidx file to construct the download link, this is not sufficient in the most general case.

closed 2022-06-07 by chaws