Open-CMSIS-Pack/cpackget #91: sha1 check during pack add

jeromecoutant, 2022-05-25:
Hi

For example, in CMIS_PACK_ROOT/ARM/CMSIS/5.9.0
there is a sha1 file: ARM.CMSIS.sha1

Does cpackget check this file while unzipping pack file?
If some sha1 is failing, is pack rejected ?

More info in https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/bash_script.html

Thx

chaws, 2022-05-26:
No, cpackget does not check that sha1 file. I wasn't aware of that too, thanks for raising this issue! 

@jkrech is this still part of packs schema? I'm asking because we never ran into that file before.

jkrech, 2022-05-30:
@chaws, this is "just a file"in the CMSIS pack but this does not mean it is part of "the specification". Therefore there is currently no requirement for cpackget to "validate" it. 
In my view we should progress the specification of digital signatures for packs and how tools like cpackget should treat this information.

chaws, 2022-05-30:
@jkrech Since this file is created in https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/bash_script.html, which is located in the official docs, cpackget could validate it only if the `*.sha1` file is present. That should check for file integrity until digital signature is implemented. What do you think?

jkrech, 2022-05-30:
I would safe the time and focus on defects and other features, as this will become obsolete and is not defined by the standard.

chaws, 2022-05-31:
Ack.

Just FYI @jeromecoutant I'm closing this issue and adding a note in https://github.com/Open-CMSIS-Pack/cpackget/issues/8 mentioning that per-file integrity check should be accomplished as well.