Open-CMSIS-Pack/cpackget #93: Help text on Windows misleading

ReinhardKeil, 2022-05-30:
**Describe the bug**
On windows the help message for -R option is somewhat confusing

Instead of:
```  -R, --pack-root string   Specifies pack root folder. Defaults to CMSIS_PACK_ROOT environment variable (default "C:\\Users\\reikei01\\AppData\\Local\\Arm\\Packs")
```

It should be: 
```  -R, --pack-root string   Specifies pack root folder. Defaults to CMSIS_PACK_ROOT environment variable (default "C:\Users\reikei01\AppData\Local\Arm\Packs")
```

**To Reproduce**
Steps to reproduce the behavior:
1. Run the cpackget on Windows 10 with the following command line
```
cpackget add -h
```

Note that this issue might exist on other places too.

------

With the command `cpackget index -h` (on Windows) there is another misleading output.

Instead of:
```
Updates the public index in CMSIS_PACK_ROOT/.Web/index.pidx using the file specified by the given url.
```

it could be
```
Updates the public index in %CMSIS_PACK_ROOT%/.Web/index.pidx using the file specified by the given url.
```

I believe the %CMSIS_PACK_ROOT% notation also works on Linux

chaws, 2022-05-30:
> `-R, --pack-root string   Specifies pack root folder. Defaults to CMSIS_PACK_ROOT environment variable (default "C:\\Users\\reikei01\\AppData\\Local\\Arm\\Packs")`

The path is being escaped. I'll look into getting the path in this and other places un-escaped as well. The same for the environment variable notation.

lud0v1c, 2022-07-05:
Closing this for now, subscribed to the Cobra issue in case any relevant updates or the pending PR is pushed as not to add such additional complexity (the first part is [already done ](https://github.com/lud0v1c/cpackget/commit/bc6cb3918a41ddceba1eda37ac65ac4823ab14b7)).

closed 2022-07-05 by lud0v1c