Open-CMSIS-Pack/cpackget #96: Question: What is the difference under removing packs

ReinhardKeil, 2022-06-07:
I have a question regarding the documentation section https://github.com/Open-CMSIS-Pack/cpackget/blob/main/README.md#removing-packs

What is the difference between 
  - removing packs (the first 2 examples)
  - removing packs that were installed via PDSC file (the last 2 examples)

Is this just an artifact of the previous command versions or is there more to it?

jkrech, 2022-06-07:
@chaws you need to update the documentation for removing pack from the `local_repository.pidx`
a) removing a pdsc entry from the `./Local/local_repository.pidx` requires to specify the `pdsc` file extension!
e.g.:
Remove pack Vendor.PackName from the local_repository.pidx:
```
    cpackget rm Vendor.PackName.pdsc
```
There is no way to specify a version in this context. If there is multiple pdsc files with the same name, the full path needs to be specified to remove them from the `local_repository.pidx`
b) don't call packs that are listed in the `local_repository.pidx` local packs, because local packs are regular read-only installed packs which are not listed in the public index.pidx. A copy of their pdsc file is stored in `.Local` directory.

chaws, 2022-06-07:
Argh, I was sure I updated the README, but it was actually in the help:

```bash
$ cpackget rm --help

Remove a pack using the reference "Vendor.Pack[.x.y.z]", "Vendor::Pack[@x.y.z]" or "[path/to/]Vendor.Pack.pdsc".

  $ cpackget rm Vendor.Pack.1.2.3
  $ cpackget rm Vendor::Pack@1.2.3
  
  Use the syntax above to let cpackget determine
  the location of pack files prior to removing them.
  
  $ cpackget rm Vendor.LocalPackInstalledViaPdsc.pdsc
  $ cpackget rm path/to/Vendor.LocalPackInstalledViaPdsc.pdsc
  
  cpackget also identifies if the pack was installed via
  PDSC file. In this case, cpackget will remove its reference
  from ".Local/local_repository.pidx".

  In the first example, since just the basename of the PDSC file
  path was specified, cpackget will remove *ALL* references of the
  PDSC file it might find. Since it is possible to have many versions
  of the same pack installed via different PDSC file paths, one may
  wish to remove a specific one by specifying a more complete
  PDSC file path, as shown in the second example.

The version "x.y.z" is optional.
Cache files (i.e. under CMSIS_PACK_ROOT/.Download/)
are *NOT* removed. If cache files need to be actually removed,
please use "--purge".
```

I'll add that to docs too.