Open-CMSIS-Pack/cpackget #97: Question: is extension *.pack supported

ReinhardKeil, 2022-06-07:
It is unclear if the extension `*.pack` is supported when specifying packs.  

I found the following inconsistency:
```
PS C:\Test\Demo_EW> cpackget rm AWS.FreeRTOS_LTS.2020.12.1-Beta.pack
I: Using pack root: "C:\Users\reikei01\AppData\Local\Arm\Packs"
I: Removing [AWS.FreeRTOS_LTS.2020.12.1-Beta.pack]
PS C:\Test\Demo_EW> cpackget add AWS.FreeRTOS_LTS.2020.12.1-Beta.pack
I: Using pack root: "C:\Users\reikei01\AppData\Local\Arm\Packs"
I: Adding pack "AWS.FreeRTOS_LTS.2020.12.1-Beta.pack"
E: File "AWS.FreeRTOS_LTS.2020.12.1-Beta.pack" does't exist
```
While the command `rm` seems to accept the extension `*.pack`, the command `add` seems not to handle it.

**Environment:**
```cpackget version 0.5.0 (C) 2022 Linaro```
Running on Windows 10.

jkrech, 2022-06-07:
@ReinhardKeil, you should not specify the `.pack` extension in this case, because cpackget is assuming that you want to install that file from the current directory:
```
cpackget add .\AWS.FreeRTOS_LTS.2020.12.1-Beta.pack
```
In this case you want to download and install a public pack and you should specify the pack ID:
```
AWS.FreeRTOS_LTS.2020.12.1-Beta
```
or matching the csolution notation
```
   AWS::FreeRTOS_LTS@2020.12.1-Beta 
```

ReinhardKeil, 2022-06-07:
I improved the documentation, but when specifying the extension explicitly, it should be assumed that the pack is on the local computer. I would remove the option, that it gets downloaded via the public index in this case.  That would make the operation clearer.

See also the documentation in https://github.com/Open-CMSIS-Pack/devtools/tree/main/tools/cpackget/docs

jkrech, 2022-06-07:
Note that both commands assume a file on your computer in the current directory:
```
cpackget add .\AWS.FreeRTOS_LTS.2020.12.1-Beta.pack
cpackget add AWS.FreeRTOS_LTS.2020.12.1-Beta.pack
```
I just added the '.\' to make it more obvious.

closed 2022-06-07 by ReinhardKeil