Open-CMSIS-Pack/cpackget #105: Extend cpackget to handle pack requirements

jkrech, 2022-06-28:
The CMSIS-Pack Specification supports the definition of pack dependencies:

[\<requirements\>\<packages\>](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/element_requirements_pg.html#element_packages)

a) extend the `list` command output to add an indicator to packs that specify package requirements
b) add a command that lists the dependencies specified by the pack specified
```
$cpackget dependencies ARM::CMSIS-Driver@1.1.0
```
c) add a command that resolves the dependencies defined by the pack specified 
```
$cpackget resolve ARM::CMSIS-Driver@1.1.0
```
d) review whether regular install command `add` automatically installs all packs required or a sequence of `add` followed by `resolve` is preferred.

Discuss whether this shall be in scope for cpackget or not?

chaws, 2022-07-05:
> $cpackget dependencies ARM::CMSIS-Driver@1.1.0

Instead of a specific command just for pack's dependencies, we could have a `details` showing some pack info, such as available releases, description, components, etc. The dependencies could be a comma-separated list with all dependencies. What do you think? 

> d) review whether regular install command add automatically installs all packs required or a sequence of add followed by resolve is preferred.

This seems a cpackget feature in my opinion. The only thing is this would have to be turned on via a flag due to backwards compatibility:
`cpackget add ARM::CMSIS` would **NOT** install dependencies and `cpackget add --dependencies/-d ARM::CMSIS` would install or check is dependencies are installed. Does that sound reasonable?

jkrech, 2022-07-07:
I would prefer cpackget to strictly stick to pack level information. Therefore available versions and descriptions are in scope, while devices, boards, examples, components, conditions, etc.  in my view are out of scope (pack content).
Please note that pack level dependencies, must not be mixed with component dependencies!
How about adding a command `cpackget info pack` on top of `add` and `rm`. This command could output the available releases as well as requirements (see below).

I like the proposal of giving the user the option whether or not to install required pack versions. 
Probably we should use the term `requirement` rather than `dependency` as this is what it is called in the package description. An alternative option would be to add a new command `resolve` which would check for unsatisfied requirements and installs them. 
We could call:
a)  without argument -> this would check the requirements for all installed packs
b) with a pack being specified -> this would check the requirements of the specified pack
and installs all required packs.

Note, that in my view it does not matter whether a requirement is resolved through a regular installed pack version or through an entry in the local_repository.pidx

lud0v1c, 2022-07-12:
Being cpackget a "pack manager", I think it makes sense to have some notion of dependencies/requirements, even more if they are a part of the Pack specification.
Why not, when doing an `add` or `rm`, prompting the user if they want to install/delete the proposed requirements? This way cpackget wouldn't force any behavior, but the user would still be able to follow the best practices.
We could have a `cpackget info pack` command like @jkrech suggested, which would show the "dependency tree" of a specific pack (or even all, but that will probably be very resource intensive).

I see the utility in the `resolve` command, but to me that sounds more like a temporary/last resort option to "migrate" the current installation to one that respects the requirements specification.

jkrech, 2022-07-12:
Note the prompting is only addressing the interactive use case. For an unattended workflow this is not an option.

PR #152 (2023-02-20): installer: add/list pack requirements
Parses a pack's `requirements` section, installing listed dependencies (only packs), and optionally list them.
All the options used for the original package are passed down to the other required packages to be installed (recursively).

Adds a `-n/--no-dependencies` flag to skip installing dependencies, and a `list required` subcommand
to view only installed packs with dependencies and their status.

[More info on this specification.](https://open-cmsis-pack.github.io/Open-CMSIS-Pack-Spec/main/html/element_requirements_pg.html#element_packages)

Closes #105 [... cut]
commit e90e0f96 (2023-02-20): installer: add/list pack requirements (#152)

Parses a pack's <requirement/packages> section, installing
listed dependencies (only packs).
All the options used for the original package are passed down
to the other required packages (recursively).

Adds a "-n/--no-dependencies" flag to skip installing
dependencies, and a "list required" subcommand
to view only installed packs with dependencies and
their status.

Closes #105.
closed 2023-02-20 by lud0v1c