Open-CMSIS-Pack/cpackget #107: [cpackget] additional command to download latest versions of all pdsc files from index.pidx (in .Web folder)

jkrech, 2022-07-07:
cpackget was intentionally designed to not rely on the presence/cache of all pdsc files listed in the index.pidx of the .Web folder.
In case cpackget is used in the context of a tool that requires to collect information from all latest public package description files, it would be desirable to have cpackget offering this functionality. E.g.
```
cpackget init <url> --all-pdsc-files
```

JonatanAntoni, 2022-07-07:
Incremental update of an existing cache not to forget. Initial download of the Keil index with all its listed pdsc files takes its time already today. Fetching each file separately adds a lot of overhead in today's index layout.

lud0v1c, 2022-07-11:
> @JonatanAntoni Please review: updating the index (https://github.com/Open-CMSIS-Pack/cpackget/blob/main/README.md#updating-the-index) By default the pdsc files present in the .Web folder will get updated if required or even removed in case the pack is no longer present in the index.pidx. So this functionality would be already covered.

Also verified this. @JonatanAntoni any behavior you would like to see implemented regarding downloading .pdsc files?

PR #111 (2022-08-14): HTTP/Download improvements
Adds network/batch download features and improves the existing ones.
This PR introduces:
* 1 command flag
    * `cpackget init --all-pdsc-files/-a`: Downloads all PDSC files listed in the initialized public index
* 2 global flags
    * `--concurrent-downloads/-C`: Enables concurrent/parallel downloads when downloading multiple files, currently only used by the new `--all-pdsc-files` command flag. Set to 5 by default, 0 to disable concurrency.
    * `--timeout/-T`: Sets a maximum timeout for all HTTP/HTTPS downloads, in seconds. Disabled by default.

Closes #107,#110.
closed 2022-08-14 by lud0v1c