Open-CMSIS-Pack/cpackget #117: File permission for generator tool

jeromecoutant, 2022-08-26:
Hi

https://github.com/Open-CMSIS-Pack/devtools/blob/main/tools/projmgr/docs/Manual/YML-Format.md#generator-proposal

Our generator could be .sh files and need to be executed with "x" permission. (especially in Linux)

Is it up to `cpackget` to change file mode during "unpack",
or `csolution` to force x permission before executing generatot command?

Thx

jkrech, 2022-08-29:
@jeromecoutant, how about retaining the 'x' permission from the zip archive. Not sure what the host platform dependencies are. We probably should add a check to PackChk to list files with 'x' permission to ensure they are set "intentionally".
@lud0v1c, you should be able set all files permissions to read-only (clear the 'w' flag) without affecting the 'x' flag right?

lud0v1c, 2022-08-29:
Yes, I think that should be possible. Will try it out.

jkrech, 2022-09-12:
From @lud0v1c response it is unclear to me whether cpackget currently overrides/clears the execute file permission. Assuming it does, I have set the label "bug".
@jeromecoutant do you think retaining the execute permission flag would be sufficient?

jeromecoutant, 2022-09-12:
Hi
In parallel of #118 workaround (changing permission of directories), I have also changes the permissions of all sh files :-)

> do you think retaining the execute permission flag would be sufficient?

It is unclear to me how can be a pack formatted with Windows, and used in a Linux system...
Maybe forcing x permission for sh scripts can be easy to implement and this will avoid some future issues?

jkrech, 2022-09-12:
@jeromecoutant, good point. Probably the zip operation needs to be done on a UNIX system. As we may have other "executables" with or without specific file extensions, just searching for file extensions may not solve searching for *.sh either.

jkrech, 2023-01-24:
I am leaning towards letting the tools invoking a generator ensure that the corresponding file has the permissions set to 'executable' rather than trying to change permissions for all files with the file extension `.sh`. 
How likely is it that a user installing a pack is different and has different permissions to the user running the generator tool?
Is it good practice to create packs on a Linux host in order to ensure that file permissions are going to be retained during `unzip`?

mdortel-stm, 2023-01-24:
Zip does not guaranty preserving file permissions, even less if zipped on Windows and unzipped on Linux.
I tried a quick test on Unix: if you explicitly call `sh ./my_script.sh` instead of `./my_script.sh`, you don't have permission issue.
If confirmed, would it be an acceptable workaround?

jeromecoutant, 2023-01-31:
> if you explicitly call `sh ./my_script.sh`

so csolution tool has to be updated ?

jkrech, 2023-01-31:
We can clearly identify *.sh is there other files? e.g. *.exe ?

lud0v1c, 2023-01-31:
> Zip does not guaranty preserving file permissions, even less if zipped on Windows and unzipped on Linux. I tried a quick test on Unix: if you explicitly call `sh ./my_script.sh` instead of `./my_script.sh`, you don't have permission issue. If confirmed, would it be an acceptable workaround?

This is correct, we can't guarantee the permissions that the zipped files have will be the same as the unpacked ones.

Has the initial reasoning for setting installed packs as read only changed? I can see the utility of the raised issue, specially in a CI setting, but in general it seems a bit unsafe to set user/group wide run permissions on unpacked executables (probably a very edge case, but an exploit in `zip` could execute a malicious script packed inside).

Perhaps implementing it as an optional flag? Something like `cpackget add -x/--executable` , or even being able to specify what extensions to set: `cpackget add -x sh,exe,bat`

closed 2023-02-08 by jkrech