Open-CMSIS-Pack/cpackget #122: cpackget init issue with the new toolbox

jeromecoutant, 2022-09-07:
With toolbox 1.0.0:
```
$ cpackget.exe --version
cpackget version 0.7.0 (C) 2022 Linaro

$ cpackget.exe init https://www.keil.com/pack/keil.pidx
...
D: Updating public index with "https://www.keil.com/pack/keil.pidx"
D: Downloading https://www.keil.com/pack/keil.pidx to C:\xxx\.Download\keil.pidx
I: Downloading keil.pidx...
D: Downloaded 17337 bytes
...
```

With toolbox 1.1.0:
```
$ cpackget.exe --version
cpackget version 0.8.0 (C) 2022 Linaro

$ cpackget.exe init https://www.keil.com/pack/keil.pidx
...
D: Updating public index with "https://www.keil.com/pack/keil.pidx"
D: Downloading https://www.keil.com/pack/keil.pidx to C:\xxx\.Download\keil.pidx
E: Get "https://www.keil.com/pack/keil.pidx": dial tcp: lookup www.keil.com: no such host
E: failed to download file
```

Looking at commit history, I think #111 has introduced an issue

Regards,

lud0v1c, 2022-09-07:
There were some small modifications on how cpackget handles HTTP GETs, but they shouldn't affect domain resolution when using a proxy. Looking into it.

lud0v1c, 2022-09-07:
@jeromecoutant is it HTTP_PROXY or HTTPS_PROXY that you are setting when the init fails?

jeromecoutant, 2022-09-07:
I got both

lud0v1c, 2022-09-07:
From my testing, setting HTTPS_PROXY pointed to an invalid SSL certificate will fail. Is your proxy using some sort of self signed certificate?

lud0v1c, 2022-09-07:
To be able do download from `https://www.keil.com/pack/keil.pidx` via a proxy, you need a valid certificate on your HTTPS_PROXY. HTTP_PROXY won't have any effect here, as it's an HTTPS GET.

jeromecoutant, 2022-09-07:
HTTPS_PROXY and HTTP_PROXY is the same, I think there is Zscaler behind it...

But my configuration is OK, as cpackget version 0.7.0 is working well :-)

jeromecoutant, 2022-09-08:
Great, it's working!
Thx

jkrech, 2022-09-12:
@jeromecoutant please give 0.8.1 a trial and close this issue, if your problem got fixed.

closed 2022-09-12 by jeromecoutant