Open-CMSIS-Pack/cpackget #131: Debug log level not working on some commands

lud0v1c, 2022-11-04:
**Describe the bug**
The `-v` flag is not showing debug verbose messages in some commands, like the recently added `checksum-` and `signature-` commands.

**To Reproduce**
`cpackget signature-create -v Pack.Vendor.1.2.3.pack -k key.pem -c cert.pem` or any valid input for the command to run.

**Expected behavior**
Debug messages should be printed.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04.1 LTS
 - Version v0.8.3

thorstendb-ARM, 2023-06-13:
Init was missing:
`PersistentPreRunE: configureInstaller`

jkrech, 2023-06-15:
I am not sure how to validate this:

`cpackget checksum-create ARM.CMSIS.5.9.0.pack -v`
does not print anything but maybe this is expected.

closed 2023-06-29 by jkrech