Open-CMSIS-Pack/cpackget #132: linux arm64 test job broken

lud0v1c, 2022-11-04:
**Describe the bug**
The `ubuntu-latest | 1.18.x | aarch64` CI test job is failing always except for when it's triggered by a regular commit.

**To Reproduce**
Any non regular commit-triggered test action, like [this one](https://github.com/Open-CMSIS-Pack/cpackget/actions/runs/3393317003/jobs/5640513624#logs).

**Expected behavior**
Should run exactly like the other test jobs.

**Additional context**
The reason seems to be that the [chosen reference for checking out](https://github.com/Open-CMSIS-Pack/cpackget/blob/06f2789ad9e2ad2bfea19772373f090a640e9699/.github/workflows/test.yml#L105) the repo seems to only work fine for commits, as it's not valid for PRs or releases.
Unfortunately the default `actions/checkout@v2` does not seem to work with the following aarch64 emulating action, so this has to be manually done.

spcaipers-arm, 2022-11-04:
Hi @lud0v1c , do you want me to investigate this issue?

lud0v1c, 2022-11-04:
Sure @spcaipers-arm, feel free to. [Here](https://docs.github.com/en/actions/learn-github-actions/environment-variables#default-environment-variables) are the reference variables, one of those should cover all git references (push, release, commit, etc).

lud0v1c, 2022-12-15:
It seems there is still some work to do on the Release workflow ([taken from v0.9.0](https://github.com/Open-CMSIS-Pack/cpackget/actions/runs/3704378258/jobs/6276932546)):
```
Run container
  WARNING: The requested image's platform (linux/arm64/v8) does not match the detected host platform (linux/amd64) and no specific platform was requested
  error: cannot update ref 'refs/heads/local_branch': trying to write non-commit object 682e224415b7fe8a631361a2deeb40510bf30819 to branch 'refs/heads/local_branch'
  From https://github.com/Open-CMSIS-Pack/cpackget
   ! [new tag]         v0.9.0     -> local_branch  (unable to update local ref)
  Error: The process '/home/runner/work/_actions/uraimo/run-on-arch-action/v2/src/run-on-arch.sh' failed with exit code 1
```
I think the problem here is that we're trying to git pull a tag:
```bash
$  git rev-list --objects --all | grep 682e224415b7fe8a631361a2deeb40510bf30819
682e224415b7fe8a631361a2deeb40510bf30819 v0.9.0
```
Which git counts as a "non commit object". Any ideas @spcaipers-arm ?

spcaipers-arm, 2022-12-16:
So, instead of:

> git fetch origin $GITHUB_REF:local_branch
> git checkout local_branch

so can you change it:

> git fetch origin $GITHUB_REF
> git checkout $GITHUB_REF

jkrech, 2023-01-24:
@lud0v1c can you please make the suggested changes?

lud0v1c, 2023-01-24:
Yes, will include this in the next PR.

lud0v1c, 2023-02-07:
@spcaipers-arm looking at the [latest PR](https://github.com/Open-CMSIS-Pack/cpackget/actions/runs/4114580009/jobs/7102133822), I think we'll need some conditional logic:
```
From https://github.com/Open-CMSIS-Pack/cpackget
   * branch            refs/pull/144/merge -> FETCH_HEAD
  error: pathspec 'refs/pull/144/merge' did not match any file(s) known to git
```
based on the context of the action being performed. Is this already implemented in a similar project?

jkrech, 2023-02-21:
@lud0v1c did this fix the remaining problem? Can this be closed?

lud0v1c, 2023-02-21:
Unfortunately, it did not: https://github.com/Open-CMSIS-Pack/cpackget/actions/runs/4219279091/jobs/7324495111

lud0v1c, 2023-02-23:
Upon confirming with @JonatanAntoni, it's indeed working - just need to let the check always run before merging the PR (as the ref is lost upon merging).

closed 2023-02-23 by lud0v1c