Open-CMSIS-Pack/cpackget #139: Problem to update a pack

christophe0606, 2022-11-29:
**Describe the bug**
I can't update my pack.

**To Reproduce**
`cpackget --version`
cpackget version 0.8.5 (C) 2022 Linaro

`cpackget update-index`
I: Using pack root: "C:\ctools_pack"
I: Updating public index
I: Downloading index.pidx 100% |███████████████████████████████████████████████████| (103/103 kB, 198.981 kB/s)
I: Updating PDSC files of installed packs referenced in index.pidx
I: ARM::V2M_MPS3_SSE_310_BSP can be upgraded from "1.0.0" to "1.1.0"

`cpackget add  ARM::V2M_MPS3_SSE_310_BSP@1.1.0`
I: Using pack root: "C:\ctools_pack"
I: Adding pack "ARM::V2M_MPS3_SSE_310_BSP@1.1.0"
E: pack version not found in the pdsc file
E: pack version not found in the pdsc file

**Expected behavior**
I was expecting the new pack to be downloaded and installed.

**Environment (please complete the following information):**
Windows 10 Enterprise 21H2

**Additional context**
My packs are installed in `C:\ctools_pack`

christophe0606, 2022-11-29:
More info: I also have a pack installation at the default location:
`C:\Users\<user>\AppData\Local\Arm\Packs`

jkrech, 2022-11-29:
@christophe0606 this is odd. Could you please share the output if you run the command line adding `-v` (verbose).

lud0v1c, 2022-11-29:
@christophe0606 in the last  `cpackget-init` you did, was the CMSIS_PACK_ROOT environment variable pointing to the default location by any chance?

christophe0606, 2022-11-29:
@lud0v1c `CMSIS_PACK_ROOT` was still pointing to `C:\ctools_pack`

brondani, 2022-11-29:
I can reproduce it with the following approach:
```
$ cpackget --version
cpackget version 0.8.5 (C) 2022 Linaro

$ export CMSIS_PACK_ROOT=/c/Temp/PacksTest

$ cpackget init https://keil.com/pack/index.pidx
I: Using pack root: "C:\Temp\PacksTest"
I: Downloading index.pidx...

$ cpackget add ARM::V2M_MPS3_SSE_310_BSP@1.0.0
I: Using pack root: "C:\Temp\PacksTest"
I: Downloading ARM.V2M_MPS3_SSE_310_BSP.pdsc...
I: Adding pack "ARM::V2M_MPS3_SSE_310_BSP@1.0.0"
I: Downloading ARM.V2M_MPS3_SSE_310_BSP.1.0.0.pack...
I: Extracting files to C:\Temp\PacksTest\ARM\V2M_MPS3_SSE_310_BSP\1.0.0...
```
Manually remove the section `<release version="1.1.0" ... </release>` from the file
`C:\Temp\PacksTest\.Web\ARM.V2M_MPS3_SSE_310_BSP.pdsc`

```
$ cpackget update-index -v
I: Using pack root: "C:\Temp\PacksTest"
[...]
D: Copying file from "C:\Temp\PacksTest\.Download\index.pidx" to "C:\Temp\PacksTest\.Web\index.pidx"
I: Updating PDSC files of installed packs referenced in index.pidx
D: Listing files and directories in "C:\Temp\PacksTest\.Web" that match ".pdsc$"
D: Checking if "C:\Temp\PacksTest\.Web\ARM.V2M_MPS3_SSE_310_BSP.pdsc" needs updating
D: Initializing PdscXML object for "C:\Temp\PacksTest\.Web\ARM.V2M_MPS3_SSE_310_BSP.pdsc"
D: Reading pdsc from file "C:\Temp\PacksTest\.Web\ARM.V2M_MPS3_SSE_310_BSP.pdsc"
D: Searching for pdsc "ARM.V2M_MPS3_SSE_310_BSP."
D: "C:\Temp\PacksTest\.Download\index.pidx" contains 1 pdsc tag(s) for "ARM.V2M_MPS3_SSE_310_BSP."
I: ARM::V2M_MPS3_SSE_310_BSP can be upgraded from "1.0.0" to "1.1.0"
D: Downloading ARM.V2M_MPS3_SSE_310_BSP.pdsc from "http://www.keil.com/pack/"
D: Downloading http://www.keil.com/pack/ARM.V2M_MPS3_SSE_310_BSP.pdsc to C:\Temp\PacksTest\.Download\ARM.V2M_MPS3_SSE_310_BSP.pdsc

$ cpackget add ARM::V2M_MPS3_SSE_310_BSP@1.1.0
I: Using pack root: "C:\Temp\PacksTest"
I: Adding pack "ARM::V2M_MPS3_SSE_310_BSP@1.1.0"
E: pack version not found in the pdsc file
E: pack version not found in the pdsc file
```

brondani, 2022-11-30:
As you can see for the command `cpackget update-index -v` the last message is:
```
D: Downloading http://www.keil.com/pack/ARM.V2M_MPS3_SSE_310_BSP.pdsc to C:\Temp\PacksTest\.Download\ARM.V2M_MPS3_SSE_310_BSP.pdsc
```
However in my environment the **.Download** directory is still empty, did the download silently failed?
I would expect the **.Web**\ARM.V2M_MPS3_SSE_310_BSP.pdsc to be updated.

commit d530351d (2022-12-15): v0.9.0 (#142)

* installer: no parallel downloads if queue is smaller

 Do not use concurrency downloading PDSC files if the set
 number of downloads would be larger than the required number.
 This avoids waitGroup sync locks while still providing
 a reasonable threshold (concurrency is set to 5 by default).

 Fixes #139.

* installer: don't repeat progress bar when extracting

Avoids printing the progress bar repeatedly when extracting,
due to looping calls of isTerminalInteractive.

Fixes #141.

* cryptography: fix signature version validation

Correctly read the "version" field in the pack' [... cut]
closed 2022-12-15 by lud0v1c