Open-CMSIS-Pack/cpackget #141: On Windows message `Extracting files` is printed to multiple lines

ReinhardKeil, 2022-12-06:
**Describe the bug**
With Windows `cmd` the message `Extracting files` is printed multiple times. 

**To Reproduce**
Steps to reproduce the behavior:
1. Run `cpackget add -f <file>` to install missing packs
2. See error in the image below
[image]

**Expected behavior**
It would be expected to see the same behavior as with the line `Downloading...`to happen.

**Environment (please complete the following information):**
 - OS: Windows 10
 - cpackget version 0.8.5

jkrech, 2022-12-06:
@ReinhardKeil could you please try to resize your window to allow longer lines to display without line breaks?
@lud0v1c could we change the design and keep the whole line for the progress bar. We need to stay within the default width of a command shell and the available characters:

`I: Extracting files to C:\Users\jkrech\AppData\Local\Arm\Packs\ARM\CMSIS\5.9.0 100% |███████████████████████████████████████████████████████████████████████████████████████████████████████████████████████| (8642/8642, 196 it/s)
`
e.g. 
`100% ███████████████████████████████████████████████| (8642/8642, 196 it/s)`

lud0v1c, 2022-12-06:
I'll have to look at the terminal module that we're using - but I'm guessing that is possible yes.

lud0v1c, 2022-12-15:
The main problem is fixed now. Regarding what you said @jkrech, are you suggesting we drop printing the log message about what's being extracted and just keep the bar? That would in fact make it more stable, as it is the log message that "breaks" the progress bar.

jkrech, 2022-12-15:
I was thinking about two lines the first one where you extract to and the other is the progress bar .
```
 I: Extracting files to C:\Users\jkrech\AppData\Local\Arm\Packs\ARM\CMSIS\5.9.0
 45% |                             | (3953/8642, 98 it/s) [37s:47s]
```

lud0v1c, 2022-12-15:
Set the bar below the message as suggested and also synced the behavior between the "Downloading" and "Extracting" actions.

commit d530351d (2022-12-15): v0.9.0 (#142)

* installer: no parallel downloads if queue is smaller

 Do not use concurrency downloading PDSC files if the set
 number of downloads would be larger than the required number.
 This avoids waitGroup sync locks while still providing
 a reasonable threshold (concurrency is set to 5 by default).

 Fixes #139.

* installer: don't repeat progress bar when extracting

Avoids printing the progress bar repeatedly when extracting,
due to looping calls of isTerminalInteractive.

Fixes #141.

* cryptography: fix signature version validation

Correctly read the "version" field in the pack' [... cut]
closed 2022-12-15 by lud0v1c