Open-CMSIS-Pack/cpackget #143: cpackget does not accept pack names starting with numbers

brondani, 2023-01-23:
**Describe the bug**
When adding a pack having a name starting with numbers, such as `32L4R9IDISCOVERY_BSP`, an error is returned.

**To Reproduce**
Steps to reproduce the behavior:
Run the cpackget with the following args:
```
$ cpackget pack add Keil::32L4R9IDISCOVERY_BSP
```
See the error:
```
I: Using pack root: "C:\Temp\packs"
E: bad pack name: it should be one of the following formats: Vendor.Pack, Vendor.Pack.x.y.z, Vendor.Pack.x.y.z.pack, or Vendor.Pack.pdsc
```
**Expected behavior**
The pack should be downloaded and installed, no error should be returned.

**Environment (please complete the following information):**
 - OS: windows
 - Version v0.9.0

**Additional context**
It seems to be related to an incorrect regular expression `[a-zA-Z][0-9a-zA-Z_\-]+` in the name pattern:
https://github.com/Open-CMSIS-Pack/cpackget/blob/v0.9.0/cmd/utils/packs.go#L19

**Additional suggestion**
When adding a pack list file as in the following example, the combined output may be misleading concerning which pack is actually failing:
```
$ cat packs.txt
ARM::CMSIS
Keil::32L4R9IDISCOVERY_BSP

$ cpackget add -af packs.txt
I: Using pack root: "C:\Temp\csolution_test\packs\repository"
I: Parsing packs urls via file packs.txt
I: Adding pack "ARM::CMSIS"
I: Extracting files to C:\Temp\csolution_test\packs\repository\ARM\CMSIS\5.9.0...
E: bad pack name: it should be one of the following formats: Vendor.Pack, Vendor.Pack.x.y.z, Vendor.Pack.x.y.z.pack, or Vendor.Pack.pdsc
E: bad pack name: it should be one of the following formats: Vendor.Pack, Vendor.Pack.x.y.z, Vendor.Pack.x.y.z.pack, or Vendor.Pack.pdsc
```
For this reason I also suggest to improve the error messages.

lud0v1c, 2023-01-23:
Yes it's very likely related to the regex used. @brondani do other toolbox tools already use a regular expression for the naming? If so, best to use the standard here.

brondani, 2023-01-23:
@lud0v1c This is the pattern in the latest CMSIS-Toolbox release 1.4.0:
https://github.com/Open-CMSIS-Pack/devtools/blob/schemas/projmgr/1.4.0/tools/projmgr/schemas/common.schema.json#L589

jkrech, 2023-01-24:
In Pack Schema the pack's `<name>` tag today is just
```
  <xs:simpleType name="RestrictedString">
    <xs:restriction base="xs:string">
      <xs:pattern value="[\-_A-Za-z0-9]+" />
    </xs:restriction>
  </xs:simpleType>
```

lud0v1c, 2023-01-26:
> In Pack Schema the pack's `<name>` tag today is just
> 
> ```
>   <xs:simpleType name="RestrictedString">
>     <xs:restriction base="xs:string">
>       <xs:pattern value="[\-_A-Za-z0-9]+" />
>     </xs:restriction>
>   </xs:simpleType>
> ```

Have to make some adaptions since the way it's currently programmed, the name is split from the vendor's regex internally, but the Spec seems to work just fine.

PR #149 (2023-02-13): installer: fix pack name/version regex
Fixes the used regular expression for a pack's name and vendor, which didn't match the current specification. Also adds a more varied selection of incorrect strings to test against.

Closes #143.
closed 2023-02-13 by jkrech