Open-CMSIS-Pack/cpackget #150: [cpackget] MacOS tests failing

lud0v1c, 2023-02-14:
**Describe the bug**
Last week when working on a `macos` GH build & test workflow, @JonatanAntoni noticed some tests were failing. Specifically, tests regarding package naming:
```
I: Using pack root: "test-list-filter-error-message"
I: Adding pack "../../testdata/integration/1.2.3/TheVendor.PublicLocalPack.1.2.3.pack"
I: Extracting files to test-list-filter-error-message/TheVendor/PublicLocalPack/1.2.3...
E: Can't move file "test-list-filter-error-message/TheVendor/PublicLocalPack/1.2.3" to "test-list-filter-error-message/TheVendor/PublicLocalPack/1.2.3.4": rename test-list-filter-error-message/TheVendor/PublicLocalPack/1.2.3 test-list-filter-error-message/TheVendor/PublicLocalPack/1.2.3.4: permission denied
--- FAIL: ExampleListInstalledPacks_filterErrorPackages (0.01s)
got:
I: Listing installed packs, filtering by "TheVendor"
E: _TheVendor::_PublicLocalPack@1.2.3 - error: vendor, pack name incorrect format
want:
I: Listing installed packs, filtering by "TheVendor"
E: _TheVendor::_PublicLocalPack@1.2.3.4 - error: vendor, pack name, pack version incorrect format
FAIL
```
At first glance, seems related to permissions. cpackget is not able to rename a file. Since the work on CMSIS_PACK_ROOT locking and unlocking was never truly tested on a native MacOS system, it's probably related to that.
Using a physical device, he was able to confirm this behavior (see specs below).

**To Reproduce**
1. Run `make test-all` on a MacOS system.

**Expected behavior**
All tests should pass.

**Environment (please complete the following information):**
- On Github's CI: [macos-12](https://github.com/actions/runner-images/blob/main/images/macos/macos-12-Readme.md)
- MacMini M1 setup: `MacOS 12.6.1 arm64`

jkrech, 2023-02-27:
The problem is caused by attempting to rename a directory which is set to read-only.
It should be safe for all platforms to make the directory `writable` before renaming.
Case a) the re-installation of the pack fails and the rename directory will be renamed back and the read-only flag should be set for the directory again
Case b) the re-installation of the pack is successful and the renamed directory (with write permission) will be deleted.

thorstendb-ARM, 2023-06-05:
fixed, see https://github.com/Open-CMSIS-Pack/cpackget/pull/177

jkrech, 2023-06-06:
Darwin tests re-enabled and now passing.

closed 2023-06-06 by jkrech