Open-CMSIS-Pack/cpackget #158: [cpackget] cpackget fails to install pack: STMicroelectronics.stm32u5xx_hal_drivers.2.0.0-alpha.1.1  (regression in version 0.9.2 vs. 0.9.0)

jkrech, 2023-02-24:
Raised here: https://github.com/Open-CMSIS-Pack/devtools/issues/772
**Describe the bug**
installing a pack with the name `STMicroelectronics.stm32u5xx_hal_drivers.2.0.0-alpha.1.1` fails with internal tool error:
```
panic: runtime error: index out of range [1] with length 1
```

**To Reproduce**
Steps to reproduce the behavior:
Hi

I got a regression using the new cpackget version.

```
$ cpackget.exe -V
cpackget version 0.9.0 (C) 2022 Linaro

$ cpackget.exe add STMicroelectronics.stm32u5xx_hal_drivers.2.0.0-alpha.1.1
I: Using pack root: "xxx\cmsis_pack_root"
I: Adding pack "STMicroelectronics.stm32u5xx_hal_drivers.2.0.0-alpha.1.1"
I: Extracting files to xxx\cmsis_pack_root\STMicroelectronics\stm32u5xx_hal_drivers\2.0.0-alpha.1.1...
```

I am now using the latest version:
```
$ cpackget.exe -V
cpackget version 0.9.2 (C) 2022-2023 Linaro

$ cpackget.exe add STMicroelectronics.stm32u5xx_hal_drivers.2.0.0-alpha.1.1
I: Using pack root: "xxx\cmsis_pack_root"
I: Adding pack "STMicroelectronics.stm32u5xx_hal_drivers.2.0.0-alpha.1.1"
I: Extracting files to xxx\cmsis_pack_root\STMicroelectronics\stm32u5xx_hal_drivers\2.0.0-alpha.1.1...
panic: runtime error: index out of range [1] with length 1

goroutine 1 [running]:
github.com/open-cmsis-pack/cpackget/cmd/installer.(*PacksInstallationType).PackIsInstalled(0xc00017b500, 0xc0001e0410)
        /home/runner/work/cpackget/cpackget/cmd/installer/root.go:959 +0x1159
github.com/open-cmsis-pack/cpackget/cmd/installer.(*PackType).loadDependencies(0xc0001e0750)
        /home/runner/work/cpackget/cpackget/cmd/installer/pack.go:573 +0x23d
github.com/open-cmsis-pack/cpackget/cmd/installer.AddPack({0xc000028480?, 0x38?}, 0x1, 0x0, 0x0, 0x0, 0x0?)
        /home/runner/work/cpackget/cpackget/cmd/installer/root.go:158 +0xadf
github.com/open-cmsis-pack/cpackget/cmd/commands.glob..func1(0x1a85a40?, {0xc000064e10?, 0x1?, 0x1?})
        /home/runner/work/cpackget/cpackget/cmd/commands/add.go:96 +0x571
github.com/spf13/cobra.(*Command).execute(0x1a85a40, {0xc000064de0, 0x1, 0x1})
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:916 +0x862
github.com/spf13/cobra.(*Command).ExecuteC(0xc000004300)
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:1044 +0x3bd
github.com/spf13/cobra.(*Command).Execute(0x1aa5a40?)
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.6.1/command.go:968 +0x19
main.main()
        /home/runner/work/cpackget/cpackget/cmd/main.go:24 +0xca
```

**Expected behavior**
A clear and concise description of what you expected to happen.

**Environment (please complete the following information):**
 - OS: [e.g. windows/linux/darwin]
 - Version [e.g. v0.0.3]

**Additional context**
Add any other context about the problem here.

jeromecoutant, 2023-02-24:
Tested with Windows

lud0v1c, 2023-02-24:
Can you please share the pack's PDSC file? It's an error reading the pack's requirements.

jeromecoutant, 2023-02-24:
```
  <requirements>
    <packages>
      <package vendor="STMicroelectronics" name="stm32u5xx_dfp" version="2.0.0-alpha.1.9"/>
      <package vendor="ARM" name="CMSIS" version="5.9.0"/>
    </packages>
  </requirements>
```

lud0v1c, 2023-02-28:
It's related to the version being used and the fact that it has a more complex patch number (`-alpha.1.9`) and its internal representation. Probably best to skip dependency installation of packs such as this for now, with the `-n` flag, until patched.

closed 2023-03-28 by thorstendb-ARM