Open-CMSIS-Pack/cpackget #166: Issue with keil.com certificate ?

fred-r-perso, 2023-03-26:
```
fredr2010@penguin:~/gh/lwfsm$ cpackget add https://keil.com/pack/ARM.CMSIS.5.9.0.pack
I: Using pack root: "/home/fredr2010/.arm/Packs"
I: Downloading ARM.CMSIS.pdsc...
I: 100% |███████████████████████████████████████████████████████████████████████████████████████████████████████████| (199/199 kB, 122 kB/s)        
I: Adding pack "https://keil.com/pack/ARM.CMSIS.5.9.0.pack"
E: Get "https://keil.com/pack/ARM.CMSIS.5.9.0.pack": tls: failed to verify certificate: x509: certificate has expired or is not yet valid: current time 2023-03-26T07:58:51+02:00 is after 2023-03-12T11:11:15Z
E: failed to download file
E: failed to download file
```

I guess the problem is not in cpakget but on the server side because of an expired certificate.

Is it possible to tell cpackget to skip the certificate verification ?

jkrech, 2023-03-27:
The problem is that the specified `url` tries to access https://keil.com/pack/... but the correct url requires a `www.keil.com`.
I am not sure why you are not leaving it to cpackget to find out the url for you.
When you initialize the CMSIS PACK ROOT folder you specify the url to the public pack index which allows cpackget to find the urls for the public packs.

fred-r-perso, 2023-03-27:
My bad indeed, I keep on using urls but it is useless, I agree.

Just one point: the reported validity issue seems anyhow to be correct seeing the dates.

I think we can close this ticket, sorry for a any inconvenience.

closed 2023-03-27 by fred-r-perso

jkrech, 2023-03-28:
The only remaining question is, whether we should have a command line option to skip the certificate validation.
In my view this should not be necessary but rather gives some confidence in the download url being trustworthy.