Open-CMSIS-Pack/cpackget #168: Installing a new pack skipping version check

jeromecoutant, 2023-03-31:
Hi

During some development phase, we could need to force to install a pack even if the version is not the latest.

I would like to avoid this kind of error:
```
$ cpackget add STMicroelectronics.xxx
...
D: Reading pdsc from file "csfBBsSQAr\STMicroelectronics.xxx.pdsc"
D: Making sure xxx is the latest release in STMicroelectronics.xxx.pdsc
E: The pack's pdsc (STMicroelectronics.xxx.pdsc) has no release tag matching version "xxx"
E: pack version not found in the pdsc file
```

Proposition is to add a new option: `--no-version-check` ?
```
$ cpackget add -h
...
Flags:
  -a, --agree-embedded-license       agrees with the embedded license of the pack
  -x, --extract-embedded-license     extracts the embedded license of the pack and aborts the installation
  -F, --force-reinstall              forces installation of an already installed pack
  -h, --help                         help for add
  -n, --no-dependencies              do not install package dependencies
  --no-version-check                 do not check release tag
  -f, --packs-list-filename string   specifies a file listing packs urls, one per line
```

Then check in https://github.com/Open-CMSIS-Pack/cpackget/blob/main/cmd/installer/pack.go#L186
is skipped.

Thx

@LMESTM @PierreLeCorre

jkrech, 2023-03-31:
@jeromecoutant, 
FYI: the new maintainer of the cpackget repo is @thorstendb-ARM 

Sorry, I do not understand your problem yet. Note that this is usually not an arbitrary version check that can be skipped, at least not, if the task is to install a public pack.

Could you maybe refine your command line, what does `xxx` include? Is it just the name of the pack or also a version?
```
cpackget add STMicroelectronics.xxx
```

```
cpackget add Keil.ARM_Compiler
```
checks whether any version of this pack is already installed. If not it installs the latest available version.
```
cpackget add Keil.ARM_Compiler.1.0.0
```
checks whether version 1.0.0 is listed in the `<releases>` section of the pdsc file in the `.Web` folder of  `CMSIS_PACK_ROOT`. If it exists it will attempt to download and install this pack version.

```
cpackget add Keil.ARM_Compiler.0.0.1
```
Version 0.0.1 does not exist in the Keil.ARM_Compiler.pdsc file in the `.Web` folder, therefore the tool will raise an error:
```
E: pack version not found in the pdsc file
```
But you seem to have another use case for getting this error message.

Does the version information in the pack file (Keil.ARM_Compiler.<version>.pack) not match the version attribute of the top entry <release> tag of the contained pdsc file (Keil.ARM_Compiler.pdsc)?

jeromecoutant, 2023-03-31:
> FYI: the new maintainer of the cpackget repo is @thorstendb-ARM

Note the assignment is automatic, I didn't choose any one :-)

About my issue, we are releasing a pack version with several steps.
At each step, pack version is a little bit different, but internal pdsc content is the same (the final version).

jkrech, 2025-04-16:
Would it be acceptable to use meta data in the pack version described in the pdsc?

`<major>.<minor>.<patch>+build1`

MyVendor.MyPack@1.0.0.pack
- MyVendor.MyPack.pdsc
  `<release version="1.0.0+build1" ...>`

Meta Data should be ignored by cpackget and the pack would be installed into folder `MyVendor/MyPack/1.0.0`

jeromecoutant, 2025-04-16:
I think we should better close the issue now
Thx

closed 2025-04-16 by jeromecoutant