Open-CMSIS-Pack/cpackget #174: Wrong default pack directory under MacOS

KeilChris, 2023-05-24:
I think that the default for the CMSIS_PACK_ROOT environment variable under MacOS in cpackget is wrong. I have tried to run the gen_pack.sh script on my new MacbookPro. I had not set the CMSIS_PACK_ROOT environment variable. And cpackget tried to search for packs in ~/.cache/arm/Packs while I had already an installation directory under ~/cache/arm/packs (last word with lower case letters at the beginning). 

Tested on MacOS 13.4.

jkrech, 2023-05-29:
@KeilChris in the second path above there is no dot in front of cache. Is that a typo or another difference?
According to this document https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/installation.md#cmsis_pack_root
the correct default path should be all lower case and with a dot in front of cache.

JonatanAntoni, 2023-06-07:
Gents, I double checked and this was a misalignment between gen-pack library and cpackget. I adopted the path to `~/.cache/arm/packs` on both, Mac and Linux, see Open-CMSIS-Pack/gen-pack#37.

KeilChris, 2023-06-12:
After deleting and reinitializing, cpackget now uses ~/.cache/arm/packs.