Open-CMSIS-Pack/cpackget #178: cpackget add with option --force-reinstall fails when pack was previously install using MDK v5

ReinhardKeil, 2023-06-06:
When the pack STM32C0xx_DFP was previously installed using the MDK pack installer, the cpackget --force-reinstall fails.

```
>cpackget add Keil::STM32C0xx_DFP --force-reinstall
I: Using pack root: ".\AppData\Local\Arm\Packs" (default mode - no specific CMSIS_PACK_ROOT chosen)
I: Adding pack "Keil::STM32C0xx_DFP"
E: Can't move file ".\AppData\Local\Arm\Packs\Keil\STM32C0xx_DFP" to ".\AppData\Local\Arm\Packs\Keil\STM32C0xx_DFP_tmp": rename .\AppData\Local\Arm\Packs\Keil\STM32C0xx_DFP .\AppData\Local\Arm\Packs\Keil\STM32C0xx_DFP_tmp: Access is denied.
```

**Question: Does cpackget correctly set the file attributes?**

Environment: Windows
cpackget version 0.9.2 (C) 2022-2023 Linaro

jkrech, 2023-06-06:
I am unable to reproduce the problem on my Windows machine but I know that we had a problem in this area for macOS.
On windows we have previously seen scenarios where a renaming of directories fails, while some anti-virus software scans the content of the directory.
@thorstendb-ARM, please try to make this reproducible and consider adding a retry in case the operation fails with some delay in between attempts.

ReinhardKeil, 2023-06-07:
Cannot reproduce anymore

closed 2023-06-07 by ReinhardKeil