Open-CMSIS-Pack/cpackget #182: [cpackget] implement potential download speedup by fetching pdsc files from keil.com

jkrech, 2023-06-15:
**Describe The Problem To Be Solved**
`cpackget init <index-url> -a` is a time consuming action as it downloads all pdsc files listed in the downloaded index.pidx file from `<index-url>`. By default the files are fetched from the origin `<url>`.  Some of the servers may not be as responsive as others.

In case of the `https://www.keil.com/pack/index.pidx` the pdsc files of the successfully imported packs are "cached" on the server and can be downloaded from `https://www.keil.com/pack`.

We should measure whether this results in a speedup for this use case.
The other benefit is that only successfully imported pack versions are present in the .Web cache  (gating faulty pack versions).

The other affected command is:
`cpackget update-index`

thorstendb-ARM, 2023-06-20:
Added check of PDSC and PublicIndexXML URL against "https://www.keil.com/pack/" and switching to that address if suitable.
Added Log message: `Switching to cache: "https://www.keil.com/pack/"`
See https://github.com/Open-CMSIS-Pack/cpackget/pull/183

thorstendb-ARM, 2023-06-20:
Please review >= 0.9.5

closed 2023-06-29 by jkrech