Open-CMSIS-Pack/cpackget #187: cpackget not applying write-protection to installed packs

JonatanAntoni, 2023-06-22:
**Describe The Problem To Be Solved**

cpackget shall use same permission restrictions to write-protect content of `CMSIS_PACK_ROOT` than used in RteModel, see Open-CMSIS-Pack/devtools#1006.

Historically, MDK PackInstaller on Windows is used to set write-protection to all files installed from a CMSIS-Pack into the pack installation folder. This is used as a hint for users not to modify the content, accidentally. `cpackget` shall continue this approach aligned with the the C++ implementation in Open-CMSIS-Pack/devtool's RteModel.

**Suggest A Solution**

`cpackget` shall
- remove all write-flags from files and folders recursively after extracting a pack into `CMSIS_PACK_ROOT` beginning at the folder level of pack version, i.e., `<CMSIS_PACK_ROOT>/<VENDOR>/<PACK>/<VERSION>`. Write-permissions of folders above the `<VERSION>` folder shall be untouched.
- restore the write-flags before doing changes to the pack folder, i.e., uninstalling or reinstalling a pack. The flags active on the parent folder (e.g., `<CMSIS_PACK_ROOT>/<VENDOR>/<PACK>`) shall be used as a reference. If the parent folder for whatever reason is write-protected as well, some reasonable defaults need to be chosen.

thorstendb-ARM, 2023-06-28:
@JonatanAntoni, can you please provide a reproducer?
Tested with Keil::STM32C0xx_DFP 1.0.0, and all files are write protected. Also the .Web folder is.

JonatanAntoni, 2023-06-28:
I can confirm that with latest cpackget 0.9.4 the installed packs are already write-protected. I still have some packs in my pack folder which are not write protected that have been installed by older version.

closed 2023-06-28 by JonatanAntoni