Open-CMSIS-Pack/cpackget #196: [cpackget] add command shall download the pack and extract the license files to `.Download`

jkrech, 2023-07-24:
**Describe The Problem To Be Solved**
If `cpackget` extracts the license files without extracting all files of the pack, an graphical UX front-end tool could directly read and display the license without the need to convert and pipe the content.

**Suggested solution**
`cpackget add <pack>` command reads the pdsc file and identifies the license files for the `<pack>` which get extracted and stored in the `./Download/<vendor>.<name>.<version>.<lic_file_name>.<lic_file_ext>`

`cpackget rm <pack>` command removes all files `./Download/<vendor>.<name>.<version>.*

**Alternative**
`cpackget add <pack>` command reads the pdsc file and identifies the license files for the `<pack>` which get extracted and stored in the `./Download/<vendor>.<name>.<version>.licenses/` directory.

`cpackget rm <pack>` command removes directory  `./Download/<vendor>.<name>.<version>.licenses/` and content recursively for all files and sub-directories.

thorstendb-ARM, 2023-08-09:
cpackget already supports license extraction:
`"args": ["add", "Keil::MDK-Middleware_Graphics", "--extract-embedded-license", "--encoded-progress"]`

Added parseable reporting format:
```
 * L: License file follows
I: [L:F"C:\Users\thode01\AppData\Local\Arm\Packs\.Download\Keil.MDK-Middleware_Graphics.1.3.0.pack.license.rtf"]
```
After extracting, the external tool can read and show the license. Then call cpackget again using:
`"--agree-embedded-license"`

thorstendb-ARM, 2023-09-05:
Using the original way for cpackget extract license, this has also been added to the MDK Pack Installer.
see cpackget v1.0.1

closed 2023-09-05 by thorstendb-ARM