Open-CMSIS-Pack/cpackget #206: [cpackget] cpackget add error message is incorrect if the specified pack cannot be found

jkrech, 2023-09-05:
**Describe the bug**
cpackget error message is specifying the wrong command and an incorrect example url

**To Reproduce**
When running the following command
```
$ cpackget add Not::AThing
I: Adding pack “Not::AThing”
E: URL for the pack cannot be determined. Please consider updating the public index. Ex: cpackget index --force https://keil.com/pack/index.pidx
```

**Expected behavior**
The command for updating the public index is:
`cpackget update-index`
The url for the default public index.pidx is: `https://www.keil.com/pack/index.pidx`

**Environment (please complete the following information):**
cpackget version 1.0.1

thorstendb-ARM, 2023-09-06:
> The command for updating the public index is:
> `cpackget update-index`
Updated:
`URL for the pack cannot be determined. Please consider updating the public index. Ex: cpackget update-index`

> The url for the default public index.pidx is: https://www.keil.com/pack/index.pidx
Checked all occurences, no issues found (this one has been deleted as "update-index" does not take a URL)

thorstendb-ARM, 2023-09-06:
Will also remove deprecated (marked) commands. Checked that they are not listed:
- https://github.com/Open-CMSIS-Pack/cmsis-toolbox/blob/main/docs/build-tools.md#cpackget-invocation
- https://github.com/Open-CMSIS-Pack/cpackget

Commands:
- 	Use:               "index <index url>",
	Short:             "Updates public index",
- 	Use:               "pack",
	Short:             "Adds/Removes Open-CMSIS-Pack packages",
- 	Use:               "pdsc",
	Short:             "Adds or removes Open-CMSIS-Pack packages in the local file system via PDSC files.",

closed 2023-09-06 by jkrech