Open-CMSIS-Pack/cpackget #229: [blocker] cpackget `crashes` when a local pack uses `<url>file:///</url>` to reference to a pack/pdsc

jkrech, 2023-12-08:
**Describe the bug**
cpackget crashes while trying to install a local pack "hosted" on the filesystem instead of a server.

**To Reproduce**
Steps to reproduce the behavior:
A `local` pack `MyVendor.Local` (not listed in `.Web/index.pidx`) installed with an updated version of the `.Local/MyVendor.Local.pdsc file with a new version (e.g. 1.1.0) specifying a `<url>file:///c:/Local_Pack_Distribution/</url> will make `cpackget` crash when running
```
  cpackget add MyVendor.Local.1.1.0
I: Adding pack "MyVendor.Local.1.1.0"
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x18 pc=0xc505b6]

goroutine 1 [running]:
github.com/open-cmsis-pack/cpackget/cmd/utils.FileExists({0xc0005211c0?, 0x3c?})
        /home/runner/work/cpackget/cpackget/cmd/utils/utils.go:226 +0x56
github.com/open-cmsis-pack/cpackget/cmd/installer.(*PackType).fetch(0xc000091ad0, 0x4?)
        /home/runner/work/cpackget/cpackget/cmd/installer/pack.go:143 +0x1cc
github.com/open-cmsis-pack/cpackget/cmd/installer.AddPack({0xc0000b2030?, 0x14?}, 0x1, 0x0, 0x0, 0x0, 0x0?)
        /home/runner/work/cpackget/cpackget/cmd/installer/root.go:116 +0x639
github.com/open-cmsis-pack/cpackget/cmd/commands.glob..func1(0xc000160400?, {0xc0002242a0?, 0x4?, 0xec989a?})
        /home/runner/work/cpackget/cpackget/cmd/commands/add.go:116 +0x5b9
github.com/spf13/cobra.(*Command).execute(0x13ea4c0, {0xc000224240, 0x3, 0x3})
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:983 +0xabc
github.com/spf13/cobra.(*Command).ExecuteC(0xc000186000)
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1115 +0x3ff
github.com/spf13/cobra.(*Command).Execute(0x100000000e60b11?)
        /home/runner/go/pkg/mod/github.com/spf13/cobra@v1.8.0/command.go:1039 +0x13
main.main()
        /home/runner/work/cpackget/cpackget/cmd/main.go:26 +0xe5

```

**Expected behavior**
Install the pack version 1.1.0 copying it from the file location specified by the  `<url>` in the pdsc file in the `.Local` pack directory.

**Environment (please complete the following information):**
 - OS: windows
 - Version 10 (22H2)

**Additional context**
Add any other context about the problem here.

jkrech, 2023-12-11:
this should fix part of: https://github.com/Open-CMSIS-Pack/devtools/issues/1242

thorstendb-ARM, 2024-01-09:
fixed crash inside FileExists(): info from os.stat was nil.

closed 2024-07-15 by jkrech