Open-CMSIS-Pack/cpackget #236: cpackget did not behave as expected in updating a certain pack

omarArm, 2024-01-08:
**Describe the bug**
On my machine I have CMSIS v5.9.0 I tried to add a new version (v6.0.0) using `cpackget add` command. The expected behaviour is that I will eventually have both versions installed in separate folders. However, what happened is the following:

1) I received an "already installed" message
2) I received a request to "force" my installation instead
3) when I forced my installation, it erased the old version (v5.9.0) and installed the newest version (I expected to have both versions installed) 

**To Reproduce**

`> cpackget add ARM::CMSIS@5.9.0
I: Adding pack "ARM::CMSIS@5.9.0"
I: Downloading ARM.CMSIS.5.9.0.pack...

I: Extracting files to /Users/omaelk01/.cache/arm/packs/ARM/CMSIS/5.9.0...
> cpackget add ARM::CMSIS  
I: Adding pack "ARM::CMSIS"
E: Pack "ARM::CMSIS" is already installed here: "/**/**/.cache/arm/packs/ARM/CMSIS", use the --force-reinstall (-F) flag to force installation`

**Expected behavior**
I expected that:
1) I won't be asked to force the installation
2) I will have both versions in two separate folders

**Environment**
 - MacOS
 - cpackget v2.0.0

jkrech, 2024-01-08:
@omarArm have you tried installing
`cpackget add ARM::CMSIS@6.0.0`

My suspicion is that cpackget will not know about 6.0.0 because your pack index file is not up-to-date and you need to run
`cpackget update-index`

Afterwards also
`cpackget add ARM::CMSIS` 
should install pack version 6.0.0 without specifying it, as it is the new "latest" version.

omarArm, 2024-01-08:
I ran `cpackget update-index` before doing a `cpackget add ARM::CMSIS`, but it didn't solve the issue. I have to clearly specify the version so that it doesn't ask for a Force, i.e., do a `cpackget add ARM::CMSIS@6.0.0` instead of relying on the tool itself to find and install the latest version by itself using a simple `cpackget add ARM::CMSIS`

jkrech, 2024-01-08:
I see. Thanks for clarifying. You are right, we don't have an explicit `command` to install the latest version of a pack under all circumstances. If you just specify the packID without version, this is interpreted as "any version". Since version 5.9.0 is already installed the version check is satisfied. Unfortunately the tool output does not really tell you anything about the version being already installed.
If version 5.9.0 would not be present, and cpackget has a choice of pack versions to choose from, it will always install the latest pack version.
I think cpackget is missing a way to specify the "latest pack version" (I have updated the issue title to reflect that).

ReinhardKeil, 2024-01-19:
Suggest to add two commands

- "update-list" that does "update-index", checks for all installed packs if there is an update available and lists packs that may be updated.
- "update" that does "update-index", checks for all installed packs if there is an update available and downloads the updated pack.  

By adding a pack name, cpackget would only update the specified pack, for example:
```cpackget update ARM::CMSIS``` would only update the ARM::CMSIS pack.

jkrech, 2024-01-23:
We already have the `list` command with the additional flags 
`--cached` 
and 
`--public`
I think it would be more consistent to add a flag e.g.
`--updates`

ReinhardKeil, 2024-01-23:
Agreed, the commands could be then:

`cpackget update [<pack>]` - update specified pack or all installed packs if not pack specified.
`cpackget list --update` - list installed packs that have a newer version - provide information about the newer version.

I believe that list commands with option `--public` and `--update` should download the file `index.pidx`.

jkrech, 2024-02-06:
@ReinhardKeil, is this a "must have" for CMSIS-Toolbox 2.3.0?

jkrech, 2024-07-03:
Note, the brief answer to the initial request is that you can run:

`cpackget add ARM::CMSIS@latest` 

to install the latest pack version found. You may want to run cpackget update-index.
We need to document this `@latest`

KeilChris, 2024-08-02:
Still, we need a command that makes the user aware that there is an updated version of a pack available. You cannot expect to run through a large project with updating all packs manually (which might be unnecessary for most of them). `cpackget list --update` is really missing.

KeilChris, 2024-08-05:
Sounds good to me!

closed 2024-09-13 by jkrech