Open-CMSIS-Pack/cpackget #246: [cpackget] exclude pdsc files from deprecated packs to be downloaded and present in the `.Web` folder

jkrech, 2024-01-19:
**Describe The Problem To Be Solved**
Any processing involving all pdsc files listed in the `index.pidx` and stored in the .Web folder are unnecessarily burdened by packs that have been deprecated. The meaning of deprecated is too fold a) this pack is no longer maintained and as other components and apis evolve, the pack looses applicability b) the vendor of devices / boards discourages to start new projects with this described hardware.

**(Optional): Suggest A Solution**
Enhance `cpackget` to ignore any `<pdsc>` element in the `.Web/index.pidx` that has the attribute `deprecated` set to a valid `DATE` when running the command
```
  cpackget init -a 
  or
  cpackget update-index
```

We should decide whether the list public packs command should include the deprecated packs or not
```
  cpackget list -p
```

I would suggest to have a dedicated flag that only shows deprecated  public packs and with `-p` not to list them.

Note: It is still possible to install a deprecate pack when specifying the packID.

ReinhardKeil, 2024-01-19:
Create a duplicate #247

jkrech, 2025-10-01:
Proposal - do not add deprecated pdsc files to `pdscCache.pidx`. Remove pdsc files from .Web for deprecated pack on `cpackget rm --purge`

closed 2026-04-09 by brondani