Open-CMSIS-Pack/cpackget #253: Pack name length limit?

Jonny-vb, 2024-02-09:
**Describe the bug**
I'm not sure if this is in cpackget or elsewhere in the chain, but it appears if the pack name is too long, the install process fails, although I can't find any reference to what this limit is.
We don't use long names for general use, but we have a CI process in place to automatically check our packs, which include doing a test creation and install of the pack. To insure there are no clashes we give the pack a unique name based off the build name, but this can be quite lengthy. A recent failure had the pack name at 108 characters (excluding the vendor name and version).

This could also just be to do with path length issues in Windows. The pack name is repeated twice in the pdsc path (folder name for pack name and then again in the pdsc file name). At 108 characters that takes my total path length to be 261 characters for the installed pdsc; over the normal limit of 256.

**Expected behavior**
Either improved documentation (ideally via the schema) to limit the pack name length, or handling of this better during the install process (some warning to the user that the install won't work)

**Environment (please complete the following information):**
Windows 10

jkrech, 2024-02-12:
I would like to narrow this down a little bit.
a) When you run the command `cbuild list environment` does it print the line:
```
  Long pathname support=enabled
```
b) What is the output for CMSIS_PACK_ROOT in the command above?
c) Am I right in assuming that you run something `cpackget add my/long/path/to/my/pack/myvendor.mypack.major.minor.patch.pack`?
Can you run the command with the flag: -v and attach the log file here? Maybe it becomes clearer where it fails?
I am afraid it will be a combination of filename and path length, therefore limiting the vendor name and pack name will not solve the issue. E.g. you could have a crazy length version qualifier after the `-` which to my knowledge is not limited.
Anyhow, do you have any suggestion for a good size limit for vendor and name? E.g. 32 characters each

Current max name: 46
Current max vendor: 19

jkrech, 2024-07-15:
Lack of additional information to properly reproduce the problem. Hence closing for the time being.

closed 2024-07-15 by jkrech