Open-CMSIS-Pack/cpackget #268: [cpackget] must strip meta data  from version

edriouk, 2024-03-07:
**Describe the bug**
Currently cpackget does not remove meta data information from versions before comparison and processing.
According to Semantic Versioning, meta-data located after '+' character should be ignored.

**To Reproduce**
Steps to reproduce the behavior:
1. Run the cpackget with the following args
`cpackget add ARM::CMSIS@6.0.0`
That installs ARM::CMSIS@6.0.0 pack or says it is already installed
2. Repeat command with the following command line: 
`cpackget add ARM::CMSIS@6.0.0+meta`
This command fails with the following output:
```
I: Adding pack "ARM::CMSIS@6.0.0+meta"
E: pack version not found in the pdsc file
```

**Expected behavior**
Executing `cpackget add ARM::CMSIS@6.0.0+meta` command should succeed and install ARM::CMSIS@6.0.0 pack or report it is already installed.

see also related problem #224

edriouk, 2024-03-18:
Fixed in PR #273

jkrech, 2024-03-19:
Validation fails.

closed 2024-03-20 by edriouk

jkrech, 2024-03-22:
Validated the build successfully: https://github.com/Open-CMSIS-Pack/cpackget/actions/runs/8355669787

1.0.0-dev0 installs into directory: `1.0.0-dev0` 
1.0.0+meta0 installs into directory: `1.0.0`