Open-CMSIS-Pack/cpackget #269: cpackget rm does not work for local packs

ReinhardKeil, 2024-03-07:
**Describe the bug**
Cannot remove a local pack, despite the fact that it is mentioned that it works in `cpackget rm --help`.

**To Reproduce**
Use https://github.com/Open-CMSIS-Pack/DFP-Pack-HandsOn and execute:

```
cpackget add ACME.ACMECM4_DFP.pdsc
cpackget rm ACME::ACMECM4_DFP@1.0.0
```

**Environment:**
 - OS: Windows
 - cpackget V2.0.1

jkrech, 2024-03-07:
The above command `cpackget add ACME.ACMECM4_DFP.pdsc` requires that the current directory is where the pdsc file is located.
Note that this command adds the pdsc file to the `$CMSIS_PACK_ROOT/.Local/local_repository.pidx` file. 
In order to remove the pdsc file from the local_repository.pidx file you also need to specify the pdsc file rather than the packID.

```
cpackget rm ACME.ACMECM4_DFP.pdsc
```

ReinhardKeil, 2024-03-08:
@jkrech I can confirm that above syntax works.  Could you change the label to "documentation" + "future improvement" (sorry have no rights).

closed 2024-09-17 by ReinhardKeil