Open-CMSIS-Pack/cpackget #272: No support for relative pack versions (supported by cprojects) and no error output when attempting to use relative pack versions

joshjennings98, 2024-03-18:
**Describe the bug**

We have issues with parsing the version of packs in the format `ARM::CMSIS@>=6.0.0`. The version format `@>=6.0.0` is [valid according to the CMSIS schema for project files](https://github.com/Open-CMSIS-Pack/devtools/blob/d3ed815177c3c1bd97c3ba914b9ee88f4034b12b/tools/projmgr/schemas/common.schema.json#L1075) but not in `cpackget`. When packs are missing we use `cpackget` to install them, however it seems like `cpackget` cannot parse this. 

Instead it **exits without printing any output** and an exit code of 255. It isn't clear what the issue is when there is no error in the output.

This makes sense when checking the output of `cpackget pack add -h` where it makes no mention of the relative version format:

```
Add a pack using the following "<pack>" specification or using packs provided by "-f <packs list>":

  $ cpackget add Vendor.Pack.1.2.3
  $ cpackget add Vendor::Pack@1.2.3

  Use the syntax above to let cpackget determine
  the location of pack files prior to installing it locally.

  $ cpackget add Vendor.Pack.1.2.3.pack

  Use this syntax if you already have a pack at hand and simply
  want to install it under pack-root folder.

  $ cpackget add path/to/Vendor.Pack.pdsc

  Use this syntax if you are installing a pack that has not
  been released yet. This will install it as a local pack and
  keep a reference in ".Local/local_repository.pidx".

The file can be a local file or a file hosted somewhere else on the Internet.
If it's hosted somewhere, cpackget will first download it then extract all pack files into "CMSIS_PACK_ROOT/<vendor>/<packName>/<x.y.z>/"
If "-f" is used, cpackget will call "cpackget pack add" on each URL specified in the <packs list> file.
```

However you need to realise the problem is coming from `cpackget` to make this connection since there is no indication from `cpackget` that anything has happened at all.

**To Reproduce**

1. Run `cpackget add pack ARM::CMSIS@>=6.0.0`.
2. See no pack installed and no output printed.

**Expected behavior**

Ideally since this relative version format is supported in the CMSIS for other tools I would expect to be able to install packs using `cpackget` in the same format.

If it is not supported then I would expect there to be some kind of output stating that a problem occurred as it is hard to find the issue if there are no logs.

**Environment (please complete the following information):**
 - OS: `Ubuntu 20.04.6 LTS`
 - Version: `cpackget version 2.0.0 (C) 2021-2023 Linaro, 2023 Arm Ltd.`

jkrech, 2024-03-18:
@joshjennings98 The root cause for not seeing any output from cpackget is the `>` operator which redirects the output into a file using the version as a filename: `1.2.3`
If you read that file you can see this error message:
```
E: bad pack name: it should be one of the following formats: Vendor.Pack, Vendor.Pack.x.y.z, Vendor.Pack.x.y.z.pack, or Vendor.Pack.pdsc
```

If you put the pack string into quotes you will see the error message:

```
cpackget add "ARM::CMSIS@>=6.0.0"
```

I don't think this format is suitable for the command line.

edriouk, 2024-03-18:
The following command line can be used instead: 

`cpackget add ARM::CMSIS@6.0.0:`

Please note the trailing colon indicating a version range with only lower limit

closed 2024-04-12 by jkrech